VoodooSMBus/VoodooRMI #152: SMBUS Synaptics SYN327C only works properly after sleep.

Darkspeed294, 2022-07-25:
hello fellow hackintoshers!

I have a problem regarding the synaptics SYN327C of the HP OMEN 15 DC1000. The problem is that there is a huge delay before it starts to send packages and registering input from the point I give input on the trackpad, and the trackpad buttons don't work properly (unless I hold the trackpad buttons for about 2,5 seconds). if the trackpad is bugged, I have to keep touching the trackpad if I want to use the trackpad buttons. otherwise, the trackpad buttons will give no input. Besides this, I am unable to tap to click.

I can kinda circumvent this problem by putting then laptop to sleep and waking it, but this is a 50/50 chance at wake that it will work properly. It is incredibly frustrating to do this multiple times and this is the only thing that keeps my hackintosh from being perfect. I have been looking and trying solutions with gestures for the last 8 months, but to no avail...

This Is what I have tried already:

- IRQ and HPET patching
- all versions of VoodooRMI+VoodooSMBUS and voodoops2 kexts 
- DSDT patching of devices VoodooRMI connects to
- removing ECenabler, SMCbattery, ssdt-bat patches
- removal of voodoops2controller kexts like mouse 
- GPI0 pinning
- trying different serial IO name configurations
- several EC-USB ssdt patches
- USB mapping
- different config.plist configurations in voodoops2 and voodoormi

**To Reproduce**

Steps to reproduce the behavior:
1. install the kexts for smbus and order them accordingly
2. install voodoops2
3. remove unwanted kexts like RMII2C and Voodooinput (voodoops2controller)
4. restart
5. laptop boots successfully but trackpad is buggy
6. log in
7. sleep laptop
8. wake after 1 minute
9. if it works, keep using. if not repeat step 5 to 7

I will add all necessary files to show you all that I know about this problem. I cannot see any difference in the ioregistryexplorer when it does and when it doesn't work. Rehabmans voodoops2 works without a problem, but no gestures. I am stuck in a position that I think the problem lies within VoodooRMI, hence why I am making an issue so late. 

If there is anything I can give to help solve this problem, feel free to ask. I will check this issue regularly.
[Debug.zip](https://github.com/VoodooSMBus/VoodooRMI/files/9184901/Debug.zip)

Darkspeed294, 2022-07-25:
[IMG_5125.webm](https://user-images.githubusercontent.com/84881650/180881180-d5f25cf1-2b0f-46cb-9aa3-1f4ee033be5f.webm)
a more detailed video to show what is happening. As you can see I am holding the trackpad to get any input from the buttons. besides this, taps are not registered at all. As you can see after the sleep, the trackpad has no delay and the buttons work as it should.

Darkspeed294, 2022-07-26:
https://github.com/Darkspeed294/HP-Omen-15-DC-1xxxx-Hackintosh-Opencore

Here is the Github if you want to look into what kind of kexts and ssdts I use. I also included my DSDT

writzx, 2023-01-03:
Just wanted to add that this is not exactly limited to SMBus trackpads. I have the SYNA328B over I2C (with VoodooI2C) and have the same issue. I am not actively looking for a solution atm because the state it is in now is good enough for my use case.

Also. the problem sometimes does appear in dual-booted Windows as well. Although Windows fixes itself after some restarts. Not sure what exactly fixes it but sometimes it stays through a few restarts before fixing itself.

1Revenger1, 2024-10-12:
I released a new version which should hopefully help with this, if you are still around and want to give it a shot. I know it's been a little while.

closed 2024-10-27 by 1Revenger1