VoodooSMBus/VoodooRMI #154: Thinkpad x1 carbon gen 6 trackpad stops working after sleep on macOS 13.1

tvannoy, 2023-01-08:
## Describe the bug
Trackpad fails to suspend, then doesn't work upon wake.

This bug appears to have surfaced after I upgraded to macOS 13.1. I'm using a 6th gen Thinkpad X1 Carbon. As far as I remember, the trackpad never stopped working after wake on macOS 12.x. This bug occurs every time I sleep the laptop on macOS 13.1.

The error occurs in `rmi_f01_suspend()`, line 389:
https://github.com/VoodooSMBus/VoodooRMI/blob/998761a22c8ada7c90e8b72f3998c2fc3ba439a5/VoodooRMI/Functions/F01.cpp#L389

I tried tracing through the call stack beyond `rmi_f01_suspend()`, but ultimately didn't get very far becuase I'm not familiar with macOS kext development. The reported error code is `-110`; as best as I could tell, this error message most likely propagated up from some `VoodooSMBus` function.

This is the call stack I think is happening: `rmi_f01_suspend` --> `RMISMBus::blockWrite()` --> `VoodooSMBusDeviceNum::writeBlockData()` --> `VoodooSMBusControllerDriver::writeBlockData()` --> `VoodooSMBusControllerDriver::transfer()` --> `command_gate->run_action()`. I suspect the error is propagating from the IOCommandGate calls, but I'm not sure. 

## System Details
- Lenovo Thinkpad X1 Carbon 6th gen (based upon [Tyler Nguyen's x1c6 config](https://github.com/tylernguyen/x1c6-hackintosh))
- macOS 13.1
- VoodooRMI: 1.3.4 Debug release
- VoodooSMBUS: bundled with 1.3.4 VoodooRMI Debug release 

I've also tried with a VoodooSMBus kext built using the latest commit in the `dev` branch: ([9226dc](https://github.com/VoodooSMBus/VoodooSMBus/tree/9226dc827e73bb0d36c7993dc1b95ae0091f2f06)). I've had the same results with the official debug releases and custom-built kexts, which include additional commits not in the released versions.

Initially, I was thinking this bug might be related to https://github.com/VoodooSMBus/VoodooSMBus/pull/41, but the VoodooSMBus kext I built included that PR.

### Kexts enabled/disabled
**Enabled**
- VoodooRMI.kext
- VoodooRMI.kext/Contents/PlugIns/VoodooInput.kext
- VoodooSMBus.kext
- VoodooRMI.kext/Contents/PlugIns/RMISMBus.kext
- VoodooPS2Controller.kext
- VoodooPS2Controller.kext/Contents/PlugIns/VoodooPS2Keyboard.kext

**Disabled**
- VoodooPS2Controller.kext/Contents/PlugIns/VoodooPS2Trackpad.kext

I've also deleted VoodooPS2Mouse, as recommended in the [VoodooSMBus README](https://github.com/VoodooSMBus/VoodooSMBus). 

## To Reproduce
Steps to reproduce the behavior:
1. Put the laptop to sleep, either by closing the lid or using the Apple menu
2. Wake the laptop from sleep

## Expected behavior
Trackpad gets suspended upon sleep, and works on wake.

## Log
Log was gathered using `sudo dmesg | grep -i vrmi` and DebugEnhancer.kext.

**Relevant section of the log after sleep:**
- 119 -- 142 is the first sleep/wake cycle
- 215 -- 264 is a second sleep/wake cycle

```
[  119.211702]: VRMI - Debug: Sleep
[  119.414268]: VRMI - Error: Failed to write sleep mode: -110.
[  119.616897]: VRMI - Error: rmi_driver_clear_irq_bits: Filaed to change enabled interrupts!
[  142.791488]: VRMI - Error: Failed to reset trackpad!
[  215.469095]: VRMI - Debug: Sleep
[  215.671858]: VRMI - Error: Failed to write sleep mode: -110.
[  215.873936]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[  215.873943]: VRMI - Error: rmi_driver_clear_irq_bits: Filaed to change enabled interrupts!
[  264.729684]: VRMI - Error: Failed to reset trackpad!
```

**Longer log excerpt:**
The rest of the log is just debug messages showing trackpad inputs
```
[    7.755830]: VRMI - Info: SMBus version 2
[    7.755979]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    7.755984]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    7.755986]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[... 62 lines cut ...]
[   42.441863]: VRMI - Debug: Finger num: 0 (valid) (913, 653) [Z: 45 WX: 1 WY: 0 FingerType: 2 Pressure : 0 Button: 0]
[   42.447091]: Sandbox apply: neagent[568] <bytes>VRMI - Debug: F11 Packet
[   42.451664]: VRMI - Debug: Finger num: 0 (valid) (946, 660) [Z: 49 WX: 0 WY: 0 FingerType: 2 Pressure : 0 Button: 0]
[   42.456693]: VRMI - Debug: F11 Packet
[   42.456705]: VRMI - Debug: Finger num: 0 (valid) (946, 660) [Z: 49 WX: 0 WY: 0 FingerType: 2 Pressure : 0 Button: 0]
[   42.465340]: Sandbox apply: WiFiVelocityAgent[569] <bytes>VRMI - Debug: F11 Packet
[   42.466459]: VRMI - Debug: Finger num: 0 (valid) (1007, 671) [Z: 55 WX: 0 WY: 0 FingerType: 2 Pressure : 0 Button: 0]
[   42.471547]: VRMI - Debug: F11 Packet
[   42.471559]: VRMI - Debug: Finger num: 0 (valid) (1007, 671) [Z: 55 WX: 0 WY: 0 FingerType: 2 Pressure : 0 Button: 0]
[   42.481234]: VRMI - Debug: F11 Packet
[   42.481248]: VRMI - Debug: Finger num: 0 (valid) (1074, 675) [Z: 56 WX: 1 WY: 0 FingerType: 2 Pressure : 0 Button: 0]
[   42.486186]: VRMI - Debug: F11 Packet
[   42.486198]: VRMI - Debug: Finger num: 0 (valid) (1074, 675) [Z: 56 WX: 1 WY: 0 FingerType: 2 Pressure : 0 Button: 0]
...
[   80.735186]: VRMI - Debug: F11 Packet
[  119.211702]: VRMI - Debug: Sleep
[  119.414268]: VRMI - Error: Failed to write sleep mode: -110.
[  119.616897]: VRMI - Error: rmi_driver_clear_irq_bits: Filaed to change enabled interrupts!
[  142.791488]: VRMI - Error: Failed to reset trackpad!
[  215.469095]: VRMI - Debug: Sleep
[  215.671858]: VRMI - Error: Failed to write sleep mode: -110.
[  215.873936]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[  215.873943]: VRMI - Error: rmi_driver_clear_irq_bits: Filaed to change enabled interrupts!
[  264.729684]: VRMI - Error: Failed to reset trackpad!
```

tvannoy, 2023-01-09:
For what it's worth, XCode is complaining about "'IOPCIDevice' is deprecated: Use PCIDriverKit
" in the VoodooSMBus project. I don't know if that could have anything to do with this bug.

imorda, 2023-09-18:
Same stuff, Lenovo X1 Yoga Gen 3 (HW is really the same as in carbon gen 6)

tvannoy, 2023-09-18:
Personally, I have moved on from my hackintosh now, but this issue was a showstopper for me. The alternative VoodooPS2Trackpad kext is painful to use. I figured this project was dead, but it appears that other discussions are happening now, so let's hope there is some progress on this. Unfortunately, I'm not familiar enough with kext development and don't have enough time to pursue fixing this bug.

1Revenger1, 2024-10-12:
If you are still around, I released a new version which hopefully helps with sleep/wake. No worries if not though, it has been a little while.

tvannoy, 2024-10-21:
I'm not on a hackintosh to test this anymore, but thanks for updating!

Feel free to close this issue.

closed 2024-10-21 by 1Revenger1