VoodooSMBus/VoodooRMI #165: VoodooRMI becomes corrupt after booting macos

Shinzef, 2023-03-26:
**Describe the bug**
VoodooRMI becomes corrupt after booting Macos before the trackpad works for a couple of secs then stops. 

[image]

**To Reproduce**
1. BootMacos
2. Shutdown
3. Boot Macos Again
4. Error appears

**Log**
No log since it didn't boot.

Shinzef, 2023-03-26:
Also some more info about the touchpad only working for a couple secs:

[    2.171747]: VRMI - Info: RMI Bus (RELEASE) Starting up!
[    2.172659]: VRMI - Info: RMII2C::TPD0 reset completed
[    2.172662]: VRMI - Error: Unable to reset
[    2.172664]: VRMI - Error: RMI initial reset failed! Continuing in spite of this
[    2.178513]: VRMI - Info: Initializing F34.
[    2.178517]: VRMI - Info: F34 not implemented
[    2.178976]: VRMI - Info: Initializing F01.
[    2.181915]: VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3336-004, fw id: 2820786
[    2.183532]: VRMI - Info: Initializing F11.
[    2.189215]: VRMI - Info: Initializing F30.
[    2.190463]: VRMI - Info: Initializing F54.
[    2.190466]: VRMI - Info: F54 not implemented
[    2.191445]: VRMI - Info: Initializing F55.
[    2.191448]: VRMI - Info: F55 not implemented

1Revenger1, 2023-03-26:
https://dortania.github.io/builds/?product=VoodooRMI&viewall=true
Do you mind trying the latest build from here?

Shinzef, 2023-03-26:
Sure later, I'm going to edit this when I do that.

1Revenger1, 2024-10-22:
If you are still around, mind giving this version a try?
https://github.com/VoodooSMBus/VoodooRMI/actions/runs/11443856445/artifacts/2083312274