VoodooSMBus/VoodooRMI #170: Lenovo T490: Erratic Mouse Pointer behavior in BootPicker

5T33Z0, 2023-06-09:
Hi. I am using VooodooSMBUS and VoodooRMI to control the TrackPad on my T490 Notebook. It's controlled via SMBUS. I am using these kexts with latest commit of OC 093 with OpenCanopy and MoisePointer support enabled:

[image]

**Issue**:

If I move the mouse cursor via the trackpad in the BootPicker, the mouse pointer jumps around erratically and unpredictably and produces "ghosting" as well. It's basically unusable in this state for selecting boot entries.

I also have a Wireless Logitech M570 mouse which can move the mouse pointer as expected without any glitches. So I am wondering if this might be related to the trackpad kexts.

**Log**
I've prepared the kexts and config as described but the generated log was empty. But I have attached an IOREG file and an OpenCore Debug Boot log. I hope this helps.

Thank you

[MBP15,4_IOREG.zip](https://github.com/VoodooSMBus/VoodooRMI/files/11703178/MBP15.4_IOREG.zip)
[opencore-2023-06-09-064902.txt](https://github.com/VoodooSMBus/VoodooRMI/files/11703186/opencore-2023-06-09-064902.txt)

1Revenger1, 2023-06-09:
I'm wondering if the trackpad doesn't get reset into a basic PS/2 mouse. I can see if I can come up with something later (likely through VoodooPS2), but I'm busy for the next couple of days.

5T33Z0, 2023-06-12:
I've also noticed that trackpad scrolling works as intended when using the official "Release" versions of off all the Voodoo kexts. Scrolling feels prettyy much like it behaves in Windows. I was using the latest nightly builds of the voodoo kext befre and the trackpad felt really unresponsive in terms of scrolling. 

After reverting to release versions, srolling feels great, but "Tap on Click" doesn't work even though it's enabled in preference and the mouse issue in the boot menu remains.

5T33Z0, 2023-06-12:
Do you mean this one: https://github.com/1Revenger1/VoodooSMBus/actions/runs/4334161505

The link expired. I can try to compile it myself.

5T33Z0, 2023-06-12:
"fatal error: too many errors emitted, stopping now [-ferror-limit=]

** BUILD FAILED **
"

Do you have a pre-compiled version somewhere?

1Revenger1, 2023-06-12:
[VoodooSMBus.kext.zip](https://github.com/VoodooSMBus/VoodooRMI/files/11726342/VoodooSMBus.kext.zip)
You were likely missing MacKernelSDK. Here is a build though.
[VoodooPS2Controller-2.3.6-DEBUG.zip](https://github.com/VoodooSMBus/VoodooRMI/files/11726513/VoodooPS2Controller-2.3.6-DEBUG.zip)
Hopefully this build of VoodooPS2 helps. This does a hardware reset on shutdown so hopefully should prevent wonkiness in the picker.

5T33Z0, 2023-06-12:
Thanks a lot.

5T33Z0, 2023-06-13:
## Test Results
I've color coded the kext in my EFI to not confuse them:

- **green 1** = VoodooPS2Controller Release v235
- **green 2** = VoodooSMBus Release v235
- **red 1** = Your VoodooPS2Controller Debug v236
- **red 2**= Your VodooSMBus v236

I've tested these possible four combinations:

Kext combo | Feature | Working|
-----------|--------|------
green1 + green2 | <ol> <li> Mouse in BootPicker <li> Tap to Click <li> Trackpad Scrolling | <ol> <li> Unusable/Erratic <li> No <li> **GOOD**
red1 + green2 | <ol> <li> Mouse in BootPicker <li> Tap to Click <li> Trackpad Scrolling | <ol><li> **YES** <li> **YES** <li> **BAD**
green1 + red2 | <ol> <li> Mouse in BootPicker <li> Tap to Click <li> Trackpad Scrolling | <ol> <li> Unusable/Erratic <li> NO <li> **GOOD**
red1 + red2 | <ol> <li> Mouse in BootPicker <li> Tap to Click <li> Trackpad Scrolling | <ol> <li> **YES** <li> **YES** <li> **BAD**

Unfortunately, with these kexts there's no combination which allows all 3 features to work properly at the same time. I hope this helps.

5T33Z0, 2023-06-13:
The log is empty again. Do I have to use the OC debug versio as well or change any `Target`settings?

1Revenger1, 2023-06-13:
Uhh, is VoodooRMI loading?
Check `kextstat` for VoodooRMI, RMISMBus, and VoodooSMBus. All three should appear.

5T33Z0, 2023-06-13:
```
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
   51   13 0                  0x2f000    0x2f000    as.vit9696.Lilu (1.6.6) 8693FE90-5FB7-3C90-B8B1-5020689EFA0B <9 7 6 3 2 1>
   52    0 0                  0x8000     0x8000     com.1Revenger1.ECEnabler (1.0.5) FB35E743-D8FB-3384-8D65-7BC6F2C8CBCC <51 9 7 6 3 2 1>
   53    2 0                  0x19000    0x19000    as.vit9696.VirtualSMC (1.3.2) 65BBE2E9-6FC4-32B4-86AB-D6A17134FDDC <51 15 9 7 6 3 1>
   55    0 0                  0x18000    0x18000    ru.usrsse2.SMCBatteryManager (1.3.2) A32BE303-F5EE-330D-9A20-867EF7A25ED6 <54 53 51 15 9 7 6 3 1>
   56    0 0                  0x82000    0x82000    as.vit9696.WhateverGreen (1.6.5) 41AEF5D8-EAD1-3D78-858F-1CC9F412CD73 <51 16 9 7 6 3 2 1>
   57    0 0                  0x15000    0x15000    as.vit9696.AppleALC (1.8.3) D5AC3A43-C89C-3628-946D-ADA78F113241 <51 16 9 7 6 3 1>
   58    0 0                  0xb000     0xb000     org.acidanthera.driver.CPUFriend (1.2.7) 3ED6D916-2A58-33C8-AC37-A3E9FF39647F <51 15 9 7 6 3 2 1>
   59    0 0                  0xb000     0xb000     com.zxystd.IntelBTPatcher (2.3.0) A7B240C1-A8C8-3594-B84C-8369BBAA7D3F <51 24 9 7 6 3 2 1>
   60    0 0                  0xa000     0xa000     as.acidanthera.BlueToolFixup (2.6.7) B0ACF935-5B68-3F63-B30C-3BB93B7C1901 <51 9 7 6 3 2 1>
   61    0 0                  0xd000     0xd000     org.acidanthera.NVMeFix (1.1.1) F0362C42-110B-3D60-97B9-8DC070ECF4C6 <51 9 7 6 3 2 1>
   63    0 0                  0xa000     0xa000     as.acidanthera.BrightnessKeys (1.0.4) 82774599-2B90-3657-8650-1B4AF9C51CA3 <62 51 15 9 7 6 3 1>
   64    0 0                  0xb000     0xb000     science.firewolf.rtsf (1.0.4) 0CB21311-6D49-3E6E-89C6-3D30AD70D296 <51 9 7 6 3 2 1>
   65    0 0                  0xa000     0xa000     as.lvs1974.DebugEnhancer (1.0.8) 05A347B1-2576-35D4-ABEB-C2E08378EF80 <51 9 7 6 3 2 1>
   71    0 0                  0x56000    0x56000    org.zhen.YogaSMC (1.5.3) 904736D1-17D6-3F51-B745-23FB3191C805 <53 51 15 9 7 6 3 1>
   74    3 0                  0x19000    0x19000    as.acidanthera.voodoo.driver.PS2Controller (2.3.6) 9CAB0A50-62BF-378A-A2AB-D923B1D947DB <15 9 7 6 3 1>
   77    0 0                  0x174000   0x174000   com.zxystd.itlwm (2.2.0) 113F3085-7721-3B88-B93E-0DA2C96B0376 <49 16 7 6 3 1>
   78    1 0                  0x18000    0x18000    de.leo-labs.VoodooSMBus (3.0) C7EA611A-7A6E-3210-800F-79585B09468C <16 7 6 3>
   83    1 0                  0x3d000    0x3d000    com.1Revenger1.VoodooRMI (1.4.0) E1287942-A6B5-3B19-9658-9B0C3E650287 <7 6 3>
   84    0 0                  0xf000     0xf000     com.1Revenger1.RMISMBus (1.0) 433136B6-5CF1-39BA-9008-77936A2C6529 <83 78 15 7 6 3>
   85    0 0                  0x30000    0x30000    com.insanelymac.IntelMausiEthernet (2.5.4) 4C804832-3A9D-3A51-B89D-98C3AF98F503 <49 16 7 6 3 1>
   88    0 0                  0xa0000    0xa0000    science.firewolf.rtsx (0.9.7) 803F199A-AF66-3699-8877-A364AEA81433 <33 24 16 6 3>
  112    0 0                  0x752000   0x752000   com.zxystd.IntelBluetoothFirmware (2.3.0) 1E34ADD4-8F23-39BB-BA0A-BFCF4173EDC4 <24 7 6 3>
  113    0 0                  0xf000     0xf000     as.acidanthera.voodoo.driver.PS2Keyboard (2.3.6) 166E4329-8627-3346-9A32-A75D9CCBCC7B <74 62 9 7 6 3 1>
  114    0 0                  0x2f000    0x2f000    as.acidanthera.voodoo.driver.PS2Trackpad (2.3.6) 1B41458C-A483-32B9-BA98-161B1D49109F <74 62 7 6 3>
  115    0 0                  0xc000     0xc000     as.acidanthera.voodoo.driver.PS2Mouse (2.3.6) CD2E26C2-FC37-3946-AF0E-C8BAAD953FD5 <74 62 7 6 3>
  122    0 0xffffff7f9a2f9000 0x2000     0x2000     com.intel.driver.EnergyDriver (3.7.0) 35E739F9-BF6C-3024-A67C-750711B3FB64 <9 7 6 3>
  179    0 0                  0x1c000    0x1c000    me.kishorprins.VoodooInput (1.1.4) 70F079D6-26C9-3002-97DE-E75100805C42 <62 7 6 3>
```

1Revenger1, 2023-06-13:
Can you double check the IO Registry to see if those kexts attach?
The device name should be `SBUS` I think.

5T33Z0, 2023-06-13:
Like so:

[image]

5T33Z0, 2023-06-13:
[image]

1Revenger1, 2023-06-13:
Ugh, the logging should be working. OpenCore debug settings doesn't have any effect here. Does `sudo dmesg` have any output at all? Double check that `msgbuf=1048576` is getting applied too. Should appear in `nvram -p | grep -i boot-args`

5T33Z0, 2023-06-13:
Enter: `nvram -p | grep -i boot-args`
Return
`boot-args	msgbuf=1048576`

Now it's working:

```
[ 1895.084250]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 1895.084263]: VRMI - Error: Failed to write sleep mode: -6.
[ 1895.088285]: VRMI - Info: Sleep
[ 1895.089497]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 1895.089501]: VRMI - Error: clearIRQs: Failed to read current IRQs, continuing to clear IRQs
[ 1895.090005]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 1895.090009]: VRMI - Error: clearIRQs: Failed to read current enabled IRQs
[ 2248.422855]: VRMI - Info: Wakeup
[ 2248.849413]: VRMI - Error: Unable to read IRQ
[ 2251.435510]: VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 2 cmd: ff]
[ 2251.460755]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2251.460763]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2251.504457]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2251.504465]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2255.505432]: VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 2 cmd: e1]
[ 2255.524605]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2255.524613]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2255.568288]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2255.568295]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2256.069356]: VRMI - Debug: F03 - PS2 Command [Send: 1 Receive: 2 cmd: e2]
[ 2256.092532]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2256.092539]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2256.136225]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2256.136231]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2256.179893]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2256.179901]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2256.223576]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2256.223583]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2256.723820]: synaptics_parse_hw_state: lastFingerCount=0 clampedFingerCount=0 clickpad=0VRMI - Error: Got [7f, 0], should be [0xAA, 0x00]! Continuing...
[ 2256.723945]: VRMI - Debug: F03 - PS2 Command [Send: 1 Receive: 0 cmd: e8]
[ 2256.747206]: synaptics_parse_hw_state: lastFingerCount=0 clampedFingerCount=0 clickpad=0VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2256.748166]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2256.781866]: synaptics_parse_hw_state: lastFingerCount=0 clampedFingerCount=0 clickpad=0VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2256.791481]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2256.829488]: synaptics_parse_hw_state: lastFingerCount=0 clampedFingerCount=0 clickpad=0VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2256.835201]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2256.877104]: synaptics_parse_hw_state: lastFingerCount=0 clampedFingerCount=0 clickpad=0VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2256.878877]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2257.379240]: VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 0 cmd: e7]
[ 2257.403056]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2257.403061]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2257.446734]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2257.446740]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2257.947294]: VRMI - Debug: F03 - PS2 Command [Send: 1 Receive: 0 cmd: f3]
[ 2257.967936]: synaptics_parse_hw_state: lastFingerCount=1 clampedFingerCount=0 clickpad=0VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2257.971356]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2258.014962]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2258.014974]: synaptics_parse_hw_state: =============NORMAL PACKET=============VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2258.051361]: synaptics_parse_hw_state: lastFingerCount=0 clampedFingerCount=0 clickpad=0VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2258.058753]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2258.098862]: synaptics_parse_hw_state: lastFingerCount=0 clampedFingerCount=0 clickpad=0VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2258.102444]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2258.603532]: VRMI - Debug: F03 - PS2 Command [Send: 0 Receive: 0 cmd: f4]
[ 2258.626744]: synaptics_parse_hw_state: =============NORMAL PACKET=============VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2258.626764]: synaptics_parse_hw_state: finger 0 -> virtual finger 0VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2258.670166]: VRMI - Error: smb_get_command_code: Failed to write mapping table data
[ 2258.670188]: VRMI - Error: F03 - Failed to write to F03 TX register (-6)
[ 2259.171114]: VRMI - Info: Finish PS2 init
```

1Revenger1, 2023-06-13:
Ugh, I forgot that the debug version of VoodooPS2 can take too long to initialize, causing VoodooRMI to not find it. I've got a PR which should fix it eventually :tm: (
So that probably means that the scrolling sucked because the PS/2 bus effectively has half the bandwidth when trying to send data for two fingers (It has to send two packets, one for each finger).
Right, let me get you a release version of VoodooPS2 and let's try this again. This time, provide the logs after tapping.

I'll edit this message with the build, I'm building on a different computer

5T33Z0, 2023-06-13:
Nice, Tapping works and scrolling works great as well. Mouse pointer still freaks out in BootPicker.

Here's the log:

```
Last login: Tue Jun 13 22:13:30 on console
stunner@MacBook-Pro-von-stunner ~ % sudo dmesg | grep -i vrmi
Password:
[    2.348245]: VRMI - Info: Found PS2 Trackpad driver! Waiting for registerService()
[    2.348386]: VRMI - Debug: Found GPIO data!
[    2.348390]: VRMI - Info: VoodooPS2Trackpad finished init, starting...
[    5.215956]: Sandbox apply: dasd[132] <bytes>Sandbox apply: nesessionmanager[288] <bytes>Sandbox: nesessionmanager(288) deny(1) ipc-posix-shm-write-create com.apple.AppleDatabaseChangedSandbox apply: sysextd[289] <bytes>Sandbox apply: coreduetd[115] <bytes>VRMI - Info: SMBus version 3
[    5.366102]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    5.366107]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    5.366110]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[    5.366112]: VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
[    5.366114]: VRMI - Debug: Config DisableWhileTypingTimeout loaded: 7d0 -> fa
[    5.366117]: VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 7d0 -> fa
[    5.366119]: VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
[    5.366121]: VRMI - Debug: Config ForceTouchType loaded: 1 -> 1
[    5.366123]: VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
[    5.366125]: VRMI - Debug: Config FingerMajorMinorDiffMax loaded: a -> a
[    5.366127]: VRMI - Debug: Config PalmRejectionWidth loaded: f -> a
[    5.366129]: VRMI - Debug: Config PalmRejectionHeight loaded: 50 -> 3c
[    5.366131]: VRMI - Debug: Config PalmRejectionTrackpointHeight loaded: 14 -> 14
[    5.366133]: VRMI - Debug: Updating Configuration
[    5.366145]: VRMI - Info: RMI Bus (DEBUG) Starting up!
[    5.366157]: VRMI - Debug: Config Clickpad loaded: 0 -> 1
[    5.366159]: VRMI - Debug: Config TrackstickButtons loaded: 1 -> 0
[    5.366162]: VRMI - Info: Recieved GPIO Data
[    5.368039]: VRMI - Debug: Function F34 - IRQs: 1 CMD Base: 0x0 CTRL Base: 0x13 DATA Base: 0x0 QRY Base: 0x29
[    5.368045]: VRMI - Info: F34 not implemented
[    5.369990]: VRMI - Debug: Function F1 - IRQs: 2 CMD Base: 0x28 CTRL Base: 0x14 DATA Base: 0x6 QRY Base: 0x31
[    5.385702]: VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3471-020, fw id: 3972349
[    5.397192]: VRMI - Debug: Function F12 - IRQs: 2 CMD Base: 0x0 CTRL Base: 0x1d DATA Base: 0xc QRY Base: 0x53
[    5.408702]: VRMI - Debug: F12 - reg: 0 reg size: 1 subpackets: 1
[    5.408712]: VRMI - Debug: F12 - reg: 1 reg size: 1 subpackets: 1
[    5.408716]: VRMI - Debug: F12 - reg: 2 reg size: 4 subpackets: 2
[    5.408720]: VRMI - Debug: F12 - reg: 3 reg size: 43 subpackets: 12
[    5.408724]: VRMI - Debug: F12 - reg: 4 reg size: 1 subpackets: 1
[    5.408728]: VRMI - Debug: F12 - reg: 5 reg size: 12 subpackets: 2
[    5.408732]: VRMI - Debug: F12 - reg: 6 reg size: 26 subpackets: 11
[    5.408737]: VRMI - Debug: F12 - reg: 7 reg size: 1 subpackets: 1
[    5.408741]: VRMI - Debug: F12 - reg: 8 reg size: 6 subpackets: 2
[    5.408746]: VRMI - Debug: F12 - reg: 9 reg size: 0 subpackets: 0
[    5.408750]: VRMI - Debug: F12 - reg: 10 reg size: 0 subpackets: 0
[    5.408755]: VRMI - Debug: F12 - reg: 13 reg size: 184 subpackets: 6
[    5.413470]: Sandbox: dasd(132) allow ipc-posix-shm-write-create com.apple.dasd.budgetsVRMI - Debug: F12 - reg: 8 reg size: 15 subpackets: 5
[    5.416725]: VRMI - Debug: F12 - reg: 9 reg size: 20 subpackets: 5
[    5.416730]: VRMI - Debug: F12 - reg: 10 reg size: 19 subpackets: 5
[    5.416735]: VRMI - Debug: F12 - reg: 11 reg size: 25 subpackets: 5
[    5.416740]: VRMI - Debug: F12 - reg: 15 reg size: 7 subpackets: 3
[    5.416745]: VRMI - Debug: F12 - reg: 19 reg size: 67 subpackets: 5
[    5.416750]: VRMI - Debug: F12 - reg: 20 reg size: 1 subpackets: 1
[    5.416754]: VRMI - Debug: F12 - reg: 22 reg size: 4 subpackets: 2
[    5.416759]: VRMI - Debug: F12 - reg: 23 reg size: 3 subpackets: 3
[    5.416763]: VRMI - Debug: F12 - reg: 28 reg size: 1 subpackets: 1
[    5.416767]: VRMI - Debug: F12 - reg: 56 reg size: 4 subpackets: 1
[    5.422941]: VRMI - Debug: F12 - reg: 1 reg size: 32 subpackets: 4
[    5.422951]: VRMI - Debug: F12 - reg: 2 reg size: 1 subpackets: 1
[    5.422956]: VRMI - Debug: F12 - reg: 4 reg size: 1 subpackets: 1
[    5.422961]: VRMI - Debug: F12 - reg: 5 reg size: 12 subpackets: 4
[    5.422965]: VRMI - Debug: F12 - reg: 9 reg size: 8 subpackets: 4
[    5.422969]: VRMI - Debug: F12 - reg: 15 reg size: 1 subpackets: 1
[    5.422974]: VRMI - Debug: F12 - reg: 34 reg size: 24 subpackets: 4
[    5.422978]: VRMI - Debug: F12 - Data packet size: 0x4f
[    5.423797]: Sandbox: dasd(132) allow file-read-data /private/var/db/DuetActivityScheduler/DuetActivitySchedulerClassC.db1 duplicate report for Sandbox: dasd(132) allow file-read-data /private/var/db/DuetActivityScheduler/DuetActivitySchedulerClassC.dbSandbox: dasd(132) allow file-write-data /private/var/db/DuetActivityScheduler/DuetActivitySchedulerClassC.dbSandbox: dasd(132) allow file-read-data /private/var/db/DuetActivityScheduler/DuetActivitySchedulerClassC.db-walSandbox: dasd(132) allow file-write-data /private/var/db/DuetActivityScheduler/DuetActivitySchedulerClassC.db-walSandbox: dasd(132) allow file-read-data /private/var/db/DuetActivityScheduler/DuetActivitySchedulerClassC.db-shmSandbox: dasd(132) allow file-write-data /private/ [... cut ...]
[   22.750645]: VRMI - Debug: Finger num: 0 (valid) (777, 352) [Z: 45 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 0]
[   22.756868]: uipc_accept: peer disconnected unp_gencnt 162VRMI - Debug: F12 Packet
[   22.764647]: VRMI - Debug: Finger num: 0 (valid) (777, 352) [Z: 15 WX: 3 WY: 3 FingerType: 2 Pressure : 0 Button: 0]
[   22.778783]: VRMI - Debug: F12 Packet
[   22.791194]: 1 duplicate report for Sandbox: triald(496) deny(1) iokit-open-user-client AppleKeyStoreUserClientSandbox: parsec-fbf(495) deny(1) file-read-data /Users/stunner/Library/Caches/GeoServicesVRMI - Debug: F12 Packet
```

5T33Z0, 2023-06-13:
Don't know why this is so long. I just booted, tapped on one file and executed the command. sorry.

5T33Z0, 2023-06-14:
After accidentally letting OCAT update all kexts I had to start over again. But this is what works now:

- VoodooPS2Controller 2.3.6 (the one you posted earlier)
- VoodooSMbus 3.0
- VoodooRMI 1.4.0 (latest Dev)

Results:

- Mouse Pointer in Bootpicker: ✅ 
- Tap to Click:  ✅
- Trackpad Scrolling:  ✅

Thank you soooo much for your help!

closed 2024-10-12 by 1Revenger1

5T33Z0, 2024-10-12:
I am sorry, to inform you, tha aall the issues I mentioned in this issue returned after you released the latest version of VoodooRMI: erratic mouse behavior on th elock screen, bad scrolling, slow mouse movement.

I am reverting back to the old version from June 23.

1Revenger1, 2024-10-12:
[VoodooPS2Controller-2.3.7-DEBUG.zip](https://github.com/user-attachments/files/17351168/VoodooPS2Controller-2.3.7-DEBUG.zip)
Does this help out? Use latest VoodooRMI/SMBus

5T33Z0, 2024-10-12:
Yes, now everything is back to normal again in Sonoma and Sequoia. Thanks

1Revenger1, 2024-10-12:
https://github.com/acidanthera/VoodooPS2/pull/65
No, will merge this in sometime soon. I messed up a little and put changelog entries in the other PR which will cause conflicts. Will merge this when I merge that other one lol.