VoodooSMBus/VoodooRMI #171: SYNA32CB support

big-mak, 2023-07-10:
Didn't see SYNA32CB in the readme, would appreciate if support was added. Thanks

1Revenger1, 2023-07-11:
The trackpad likely works if it works in Linux, Have you tested VoodooRMI with this trackpad?

big-mak, 2023-07-11:
I have tested it and I can't get it to work, boot gets stuck at messages like this: `ApplePS2ALPSGlidePoint: identify: not an ALPS device: Error getting E6 report`
I have tried with and without VoodooI2C and various combinations of the relevant sattelite kexts. Discussing this [here](https://forum.amd-osx.com/threads/nootedred-renoir-cezanne-apus-installing-on-ryzen-7-5825u-installer-freezes-at-12min-every-time.4555/page-2#post) and latest EFI attached 
[EFI.zip](https://github.com/VoodooSMBus/VoodooRMI/files/12012362/EFI.zip)

1Revenger1, 2023-07-11:
Are you able to boot into macOS at all even without VoodooI2C and VoodooRMI?
If you have an I2C trackpad, you should probably disable VoodooPS2Mouse and Trackpad.

big-mak, 2023-07-13:
Used a VoodooI2C build from NootInc and it works now.

closed 2023-07-13 by big-mak