VoodooSMBus/VoodooRMI #173: [X1C6] Thumb/Joint is not detected properly -> Click+drag not working

davidohne, 2023-09-01:
**Describe the bug**
Thumb/Joint is not recognized which makes it impossible to click and drag thinks or highlight things.

**To Reproduce**
Download FingerMgmt and try to use the joint of the thumb as input.

**Expected behavior**
Click and thumb/joint is detected properly so Click+drag and highlightning of text is possible.

**Video**
Tried to sho the behavior on the X1C6 in a video: https://youtu.be/hufQXLozEl8
On a T480S it looks like this: https://youtube.com/shorts/ZnPeELd4C1g

davidohne, 2023-09-02:
I made some more observations: In Windows 10 the touchpad isn't recognizing touch input by the joint/thumb aswell. BUT: When clicking with the thumb it is recognized properly and highlighting of text and drag and drop is working fine.

Could it be that this is because the trackpad is the NFC version? I disconnected the NFC module from the touchpad but drag and drop is still not working/the same behaviour. When disconnecting the NFC module the trackpad isn't recognized as trackpad in the macOS preferences anymore - only Mouse settings are avialable which are there for the Trackpoint if I remember right?

1Revenger1, 2023-09-02:
This is mostly going to be a software issue. VoodooRMI rejects touches which are too tall/wide and doesn't revalidate them if you click with the touchpad. I believe there is some configuration you can change in the #readme which should help.

I'm busy for the next few days but I'll see when I can take a look at this.

davidohne, 2023-09-13:
I already tried several settings but it doesnt work. It's kind of funny because it only happens when highlightning text and dragging folders. When moving windows around this behaviour doesnt happen.. Can I share more information which could help in this case?

davidohne, 2023-09-13:
Logs look like this:
```
david@MacBook-Pro ~ % sudo dmesg | grep VRMI
[    2.420707]: VRMI - Info: Found PS2 Trackpad driver! Waiting for registerService()
[    2.420926]: VRMI - Debug: Too early notification - No VoodooInput on PS2 yet
[    3.146442]: VRMI - Debug: Found GPIO data!
[    3.146446]: VRMI - Info: VoodooPS2Trackpad finished init, starting...
[    5.663686]: VRMI - Info: SMBus version 3
[    5.663849]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    5.663869]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    5.663871]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[    5.663873]: VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
[    5.663876]: VRMI - Debug: Config DisableWhileTypingTimeout loaded: 7d0 -> fa
[    5.663878]: VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 7d0 -> fa
[    5.663880]: VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
[    5.663882]: VRMI - Debug: Config ForceTouchType loaded: 1 -> 1
[    5.663884]: VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
[    5.663886]: VRMI - Debug: Config FingerMajorMinorDiffMax loaded: a -> a
[    5.663888]: VRMI - Debug: Config PalmRejectionWidth loaded: f -> a
[    5.663890]: VRMI - Debug: Config PalmRejectionHeight loaded: 50 -> 3c
[    5.663892]: VRMI - Debug: Config PalmRejectionTrackpointHeight loaded: 14 -> 14
[    5.663894]: VRMI - Debug: Updating Configuration
[    5.663903]: VRMI - Info: RMI Bus (DEBUG) Starting up!
[    5.663905]: VRMI - Debug: Config Clickpad loaded: 0 -> 1
[    5.663907]: VRMI - Debug: Config TrackstickButtons loaded: 1 -> 0
[    5.663910]: VRMI - Info: Recieved GPIO Data
[    5.670189]: VRMI - Debug: rmi_probe_interrupts: Counting IRQs
[    5.698410]: VRMI - Debug: IRQ Count: 9
[    5.698448]: VRMI - Debug: rmi_init_functions: Creating functions
[    5.700532]: VRMI - Info: Initializing F34.
[    5.700536]: VRMI - Debug: Function F34 - IRQs: 1 CMD Base: 0 CTRL Base: 15 DATA Base: 0 QRY Base: 40 VER: 1
[    5.700539]: VRMI - Info: F34 not implemented
[    5.702594]: VRMI - Info: Initializing F01.
[    5.702598]: VRMI - Debug: Function F1 - IRQs: 2 CMD Base: 39 CTRL Base: 19 DATA Base: 4 QRY Base: 44 VER: 0
[    5.719398]: VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3288-011, fw id: 2812761
[    5.727712]: VRMI - Info: Initializing F12.
[    5.727716]: VRMI - Debug: Function F12 - IRQs: 2 CMD Base: 0 CTRL Base: 27 DATA Base: 9 QRY Base: 75 VER: 0
[    5.740005]: VRMI - Debug: F12 - reg: 0 reg size: 1 subpackets: 1
[    5.740009]: VRMI - Debug: F12 - reg: 1 reg size: 1 subpackets: 1
[    5.740012]: VRMI - Debug: F12 - reg: 2 reg size: 5 subpackets: 2
[    5.740014]: VRMI - Debug: F12 - reg: 3 reg size: 44 subpackets: 13
[    5.740017]: VRMI - Debug: F12 - reg: 4 reg size: 1 subpackets: 1
[    5.740019]: VRMI - Debug: F12 - reg: 5 reg size: 11 subpackets: 2
[    5.740022]: VRMI - Debug: F12 - reg: 6 reg size: 27 subpackets: 12
[    5.740024]: VRMI - Debug: F12 - reg: 7 reg size: 1 subpackets: 1
[    5.740026]: VRMI - Debug: F12 - reg: 8 reg size: 6 subpackets: 2
[    5.740044]: VRMI - Debug: F12 - reg: 9 reg size: 0 subpackets: 0
[    5.740046]: VRMI - Debug: F12 - reg: 10 reg size: 0 subpackets: 0
[    5.740048]: VRMI - Debug: F12 - reg: 13 reg size: 0 subpackets: 0
[    5.740050]: VRMI - Debug: F12 - reg: 21 reg size: 0 subpackets: 0
[    5.748315]: VRMI - Debug: F12 - reg: 8 reg size: 14 subpackets: 4
[    5.748320]: VRMI - Debug: F12 - reg: 9 reg size: 15 subpackets: 3
[    5.748322]: VRMI - Debug: F12 - reg: 10 reg size: 18 subpackets: 4
[    5.748324]: VRMI - Debug: F12 - reg: 11 reg size: 9 subpackets: 3
[    5.748326]: VRMI - Debug: F12 - reg: 15 reg size: 7 subpackets: 3
[    5.748329]: VRMI - Debug: F12 - reg: 19 reg size: 67 subpackets: 5
[    5.748331]: VRMI - Debug: F12 - reg: 20 reg size: 1 subpackets: 1
[    5.748333]: VRMI - Debug: F12 - reg: 21 reg size: 2 subpackets: 1
[    5.748335]: VRMI - Debug: F12 - reg: 22 reg size: 1 subpackets: 1
[    5.748337]: VRMI - Debug: F12 - reg: 23 reg size: 2 subpackets: 2
[    5.748338]: VRMI - Debug: F12 - reg: 28 reg size: 1 subpackets: 1
[    5.748341]: VRMI - Debug: F12 - reg: 71 reg size: 1 subpackets: 1
[    5.754604]: VRMI - Debug: F12 - reg: 1 reg size: 32 subpackets: 4
[    5.754608]: VRMI - Debug: F12 - reg: 2 reg size: 1 subpackets: 1
[    5.754610]: VRMI - Debug: F12 - reg: 5 reg size: 12 subpackets: 4
[    5.754613]: VRMI - Debug: F12 - reg: 9 reg size: 8 subpackets: 4
[    5.754615]: VRMI - Debug: F12 - reg: 15 reg size: 1 subpackets: 1
[    5.754617]: VRMI - Debug: F12 - reg: 34 reg size: 24 subpackets: 4
[    5.754619]: VRMI - Debug: F12 - Data packet size: 0x4e
[    5.757487]: VRMI - Debug: F12 - Number of fingers 4
[... 146 lines cut ...]
[   23.890691]: VRMI - Debug: Finger num: 0 (valid) (423, 427) [Z: 68 WX: 1 WY: 2 FingerType: 2 Pressure : 0 Button: 0]
[   23.905718]: VRMI - Debug: F12 Packet
[   23.905735]: VRMI - Debug: Finger num: 0 (valid) (423, 427) [Z: 68 WX: 1 WY: 2 FingerType: 2 Pressure : 0 Button: 0]
[   23.921274]: VRMI - Debug: F12 Packet
[   23.921305]: VRMI - Debug: Finger num: 0 (valid) (423, 427) [Z: 69 WX: 1 WY: 2 FingerType: 2 Pressure : 0 Button: 0]
[   23.936737]: VRMI - Debug: F12 Packet
[   23.936769]: VRMI - Debug: Finger num: 0 (valid) (423, 427) [Z: 69 WX: 1 WY: 2 FingerType: 2 Pressure : 0 Button: 0]
[   23.952085]: VRMI - Debug: F12 Packet
[... cut ...]
[   25.776322]: VRMI - Debug: Finger num: 0 (valid) (356, 451) [Z: 62 WX: 1 WY: 1 FingerType: 0 Pressure : 0 Button: 0]
[   25.791689]: VRMI - Debug: F12 Packet
[   25.791699]: VRMI - Debug: Finger num: 0 (valid) (356, 451) [Z: 64 WX: 1 WY: 1 FingerType: 2 Pressure : 0 Button: 0]
[   25.807043]: VRMI - Debug: F12 Packet
[   25.807051]: VRMI - Debug: Finger num: 0 (valid) (370, 448) [Z: 69 WX: 1 WY: 2 FingerType: 2 Pressure : 0 Button: 0]
[   25.822550]: VRMI - Debug: F12 Packet
[   25.822560]: VRMI - Debug: Finger num: 0 (valid) (388, 456) [Z: 71 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 0]```

davidohne, 2023-09-13:
IOReg: https://ufile.io/annrpvmx

davidohne, 2023-09-13:
Pressing in the down-middle of the touchpad with my thumb this is what the log shows:
```
[  900.813682]: VRMI - Debug: Finger num: 0 (valid) (616, 118) [Z: 99 WX: 3 WY: 5 FingerType: 2 Pressure : 0 Button: 0]
[  900.828888]: VRMI - Debug: F12 Packet
[  900.844075]: VRMI - Debug: F12 Packet
[  900.845102]: VRMI - Debug: Button 2 Key 1 is Down
[  900.859279]: VRMI - Debug: F12 Packet
[  900.860312]: VRMI - Debug: Button 2 Key 1 is Down
[  900.875294]: VRMI - Debug: F12 Packet
[  900.876341]: VRMI - Debug: Button 2 Key 1 is Down
[  900.890565]: VRMI - Debug: F12 Packet
[  900.891601]: VRMI - Debug: Button 2 Key 1 is Down
[  900.906003]: VRMI - Debug: F12 Packet
[  900.907039]: VRMI - Debug: Button 2 Key 1 is Down
[  900.921238]: VRMI - Debug: F12 Packet
[  900.922282]: VRMI - Debug: Button 2 Key 1 is Down
[  900.936497]: VRMI - Debug: F12 Packet
[  900.937455]: VRMI - Debug: Button 2 Key 1 is Down
[  900.951954]: VRMI - Debug: F12 Packet
[  900.952987]: VRMI - Debug: Button 2 Key 1 is Down
[  900.967443]: VRMI - Debug: F12 Packet
[  900.968477]: VRMI - Debug: Button 2 Key 1 is Down
[  900.982689]: VRMI - Debug: F12 Packet
[  900.983649]: VRMI - Debug: Button 2 Key 1 is Up
[  900.998130]: VRMI - Debug: F12 Packet
[  900.999030]: VRMI - Debug: Button 2 Key 1 is Up
[  901.012448]: VRMI - Debug: F12 Packet
[  901.336690]: VRMI - Debug: F12 Packet```

When pressing with my forefinger in the down-middle it logs:
```
[ 1111.161793]: VRMI - Debug: Velocity: 0 0 Zone: 0
[ 1111.161802]: VRMI - Debug: Finger num: 0 (valid) (643, 30) [Z: 41 WX: 0 WY: 4 FingerType: 0 Pressure : 0 Button: 0]
[ 1111.177150]: VRMI - Debug: F12 Packet
[ 1111.177180]: VRMI - Debug: Finger num: 0 (valid) (643, 30) [Z: 59 WX: 1 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[ 1111.192535]: VRMI - Debug: F12 Packet
[ 1111.192566]: VRMI - Debug: Finger num: 0 (valid) (645, 34) [Z: 70 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[ 1111.207871]: VRMI - Debug: F12 Packet
[ 1111.207901]: VRMI - Debug: Finger num: 0 (valid) (646, 36) [Z: 72 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[ 1111.223326]: VRMI - Debug: F12 Packet
[ 1111.223357]: VRMI - Debug: Finger num: 0 (valid) (646, 38) [Z: 73 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[ 1111.238309]: VRMI - Debug: F12 Packet
[ 1111.238338]: VRMI - Debug: Finger num: 0 (valid) (645, 39) [Z: 73 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[ 1111.239325]: VRMI - Debug: Button 2 Key 1 is Down
[ 1111.254043]: VRMI - Debug: F12 Packet
[ 1111.254073]: VRMI - Debug: Finger num: 0 (valid) (644, 40) [Z: 74 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 1]
[ 1111.254951]: VRMI - Debug: Button 2 Key 1 is Down
[ 1111.269089]: VRMI - Debug: F12 Packet
[ 1111.269119]: VRMI - Debug: Finger num: 0 (valid) (644, 40) [Z: 74 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 1]
[ 1111.270106]: VRMI - Debug: Button 2 Key 1 is Down
[ 1111.284290]: VRMI - Debug: F12 Packet
[ 1111.284318]: VRMI - Debug: Finger num: 0 (valid) (644, 40) [Z: 74 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 1]
[ 1111.285214]: VRMI - Debug: Button 2 Key 1 is Down
[ 1111.300171]: VRMI - Debug: F12 Packet
[ 1111.300201]: VRMI - Debug: Finger num: 0 (valid) (644, 40) [Z: 75 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 1]
[ 1111.301184]: VRMI - Debug: Button 2 Key 1 is Down
[ 1111.315523]: VRMI - Debug: F12 Packet
[ 1111.315553]: VRMI - Debug: Finger num: 0 (valid) (644, 40) [Z: 75 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 1]
[ 1111.316554]: VRMI - Debug: Button 2 Key 1 is Down
[ 1111.330825]: VRMI - Debug: F12 Packet
[ 1111.330854]: VRMI - Debug: Finger num: 0 (valid) (644, 40) [Z: 74 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 1]
[ 1111.331835]: VRMI - Debug: Button 2 Key 1 is Down
[ 1111.345789]: VRMI - Debug: F12 Packet
[ 1111.345819]: VRMI - Debug: Finger num: 0 (valid) (644, 41) [Z: 74 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 1]
[ 1111.346801]: VRMI - Debug: Button 2 Key 1 is Down
[ 1111.361611]: VRMI - Debug: F12 Packet
[ 1111.361641]: VRMI - Debug: Finger num: 0 (valid) (644, 41) [Z: 73 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 1]
[ 1111.362622]: VRMI - Debug: Button 2 Key 1 is Up
[ 1111.376802]: VRMI - Debug: F12 Packet
[ 1111.376832]: VRMI - Debug: Finger num: 0 (valid) (645, 41) [Z: 70 WX: 2 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[... 9 lines cut ...]
[ 1112.022830]: VRMI - Debug: Finger num: 0 (valid) (648, 81) [Z: 72 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 0]
[ 1112.038161]: VRMI - Debug: F12 Packet
[ 1112.038192]: VRMI - Debug: Finger num: 0 (valid) (648, 81) [Z: 74 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 0]
[ 1112.053479]: VRMI - Debug: F12 Packet
[ 1112.053508]: VRMI - Debug: Finger num: 0 (valid) (648, 81) [Z: 75 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 0]
[ 1112.054494]: VRMI - Debug: Button 2 Key 1 is Down
[ 1112.069050]: VRMI - Debug: F12 Packet
[ 1112.069080]: VRMI - Debug: Finger num: 0 (valid) (648, 81) [Z: 75 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 1]
[ 1112.069973]: VRMI - Debug: Button 2 Key 1 is Down
[ 1112.084035]: VRMI - Debug: F12 Packet
[ 1112.084066]: VRMI - Debug: Finger num: 0 (valid) (648, 81) [Z: 76 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 1]
[ 1112.085050]: VRMI - Debug: Button 2 Key 1 is Down
[ 1112.099366]: VRMI - Debug: F12 Packet
[ 1112.099400]: VRMI - Debug: Finger num: 0 (valid) (648, 81) [Z: 76 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 1]
[ 1112.100378]: VRMI - Debug: Button 2 Key 1 is Down
[... cut ...]
[ 1112.114860]: VRMI - Debug: Button 2 Key 1 is Down
[ 1112.129237]: VRMI - Debug: F12 Packet
[ 1112.129247]: VRMI - Debug: Finger num: 0 (valid) (648, 81) [Z: 76 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 1]
[ 1112.129926]: VRMI - Debug: Button 2 Key 1 is Down
[ 1112.144473]: VRMI - Debug: F12 Packet
[ 1112.144481]: VRMI - Debug: Finger num: 0 (valid) (648, 81) [Z: 76 WX: 2 WY: 2 FingerType: 2 Pressure : 0 Button: 1]
[ 1112.145180]: VRMI - Debug: Button 2 Key 1 is Down
[ 1112.159943]: VRMI - Debug: F12 Packet```

davidohne, 2023-09-13:
Here is my config, perhaps it helps aswell. All together, the trackpad comparing to the T480s which is sitting next to my x1 carbon 6th works very bad (e.g. double tap with two fingers isn't recognized properly). I'm at some point where I'm willing to change the trackpad to the non-nfc version and hope this one works better... But perhaps the logs and my configuration will show the problem, I dont understand it :(

[config.plist.zip](https://github.com/VoodooSMBus/VoodooRMI/files/12599820/config.plist.zip)

davidohne, 2023-09-13:
Another investigation is, when I'm using my thumb to move the crusor this is what the logs tell me. It looks like there is a conflict as soon at the thumb is detected?
```
[  218.277739]: VRMI - Debug: F12 Packet
[  218.293103]: VRMI - Debug: F12 Packet
[  218.308517]: VRMI - Debug: F12 Packet
[  218.323868]: VRMI - Debug: F12 Packet
[  218.339260]: VRMI - Debug: F12 Packet
[  218.354648]: VRMI - Debug: F12 Packet
[  218.370092]: VRMI - Debug: F12 Packet
[  218.385471]: VRMI - Debug: F12 Packet
[  218.400761]: VRMI - Debug: F12 Packet
[  218.415982]: VRMI - Debug: F12 Packet
[  218.431557]: VRMI - Debug: F12 Packet
[  218.447125]: VRMI - Debug: F12 Packet
[  218.462510]: VRMI - Debug: F12 Packet
[  218.477844]: VRMI - Debug: F12 Packet
[  218.493034]: VRMI - Debug: F12 Packet
[  218.508511]: VRMI - Debug: F12 Packet
[  218.523954]: VRMI - Debug: F12 Packet
[  218.539461]: VRMI - Debug: F12 Packet
[  218.554801]: VRMI - Debug: F12 Packet
[  218.570146]: VRMI - Debug: F12 Packet
[  218.585103]: VRMI - Debug: F12 Packet
[  218.600906]: VRMI - Debug: F12 Packet
[  218.616243]: VRMI - Debug: F12 Packet
[  218.631826]: VRMI - Debug: F12 Packet
[  218.647066]: VRMI - Debug: F12 Packet
[  218.662594]: VRMI - Debug: F12 Packet
[  218.677957]: VRMI - Debug: F12 Packet
[  218.693371]: VRMI - Debug: F12 Packet
[  218.708682]: VRMI - Debug: F12 Packet
[  218.724090]: VRMI - Debug: F12 Packet
[  218.739565]: VRMI - Debug: F12 Packet
[  218.754880]: VRMI - Debug: F12 Packet
[  218.770310]: VRMI - Debug: F12 Packet
[  218.785675]: VRMI - Debug: F12 Packet
[  218.801051]: VRMI - Debug: F12 Packet
[  218.816368]: VRMI - Debug: F12 Packet
[  218.831874]: VRMI - Debug: F12 Packet
[  218.847173]: VRMI - Debug: F12 Packet
[  218.862442]: VRMI - Debug: F12 Packet
[  218.877993]: VRMI - Debug: F12 Packet
[  218.893465]: VRMI - Debug: F12 Packet
[  218.908736]: VRMI - Debug: F12 Packet
[  218.923932]: VRMI - Debug: F12 Packet
[  218.939362]: VRMI - Debug: F12 Packet
[  218.953563]: VRMI - Debug: F12 Packet