VoodooSMBus/VoodooRMI #175: Tap to click not working normally

xfwdrev, 2023-10-07:
Tap to click does work normally. Have to tap the touchpad a couple of times and tap hard to register the clicks.

Tap to click also doesn't work when tapping in the middle of the trackpad. It only works when tapping in bottom center of the touchpad. 

Gesture works normally detecting up to five fingers.

My touchpad is SYN1B7F running in polling mode.

TotallyNotInUse, 2024-07-21:
I had the same problem, however I discovered that my trackpad (Thinkpad T15 Gen 1) should actually use VoodooI2C instead of VoodooRMI so I'm sharing this in hopes it helps to solve your problem.

speedie1337, 2024-08-11:
Can confirm this issue. Works sometimes when waking from sleep.

1Revenger1, 2024-10-13:
I believe should be solved by https://github.com/VoodooSMBus/VoodooRMI/actions/runs/11307575354/artifacts/2048920273 and other commits to VoodooSMBus/VoodooRMI.
Let me know if it is not.

closed 2024-10-13 by 1Revenger1