VoodooSMBus/VoodooRMI #178: No support for SYNA30B3

Thisisauser6443, 2024-05-28:
**Describe the bug**
The trackpad does work with the kext added and enabled, but the physical mouse buttons and the trackpoint remain not working. Also, enabling `VoodooRMI.kext/Contents/PlugIns/VoodooInput.kext` causes a kernel panic, similar to what happens if I try and enable the same for `VoodooI2C`. (If it helps, this is running on a HP EliteBook 840 G7)

**To Reproduce**
Steps to reproduce the behavior:
1. Add `VoodooRMI.kext` to `EFI/OC/Kexts` and add to the `config.plist`
2. Reboot and observe a kernel panic with `VoodooRMI.kext/Contents/PlugIns/VoodooInput.kext` enabled
3. Disable the aforementioned entry, and reboot again
4. Successfully boot to the userspace with a working trackpad, but with no trackpoint/mouse buttons

**Expected behavior**
Not much, I guess? I mean, I was hoping I’d get lucky, and this would work right out of the box, but as I expected, the trackpad does work (But only because of VoodooI2CHID?), but not the physical mouse buttons or trackpoint

**Log**
[log.txt](https://github.com/VoodooSMBus/VoodooRMI/files/15475135/log.txt)

Here are all of my kernel extensions, just in case you spot something off
[image]

1Revenger1, 2024-05-28:
Is the log empty? It isn't showing up on my phone although could just be my phone being weird.
The panics from VoodooInput are expected when multiple copies are injected. You only can inject one copy of any kext at a time as otherwise macOS will panic when trying to load them. Generally you only inject VoodooInput from whichever kext is responsible for the touchpad.

Thisisauser6443, 2024-05-28:
> Is the log empty? It isn't showing up on my phone although could just be my phone being weird.

Shoot, it’s empty. Bit weird, since I did follow the guide on getting logs to the letter (At least I think I did)

> Generally you only inject VoodooInput from whichever kext is responsible for the touchpad

Problem is that if I enable _either_ one of the `VoodooInput` entries, it goes to a KP

1Revenger1, 2024-05-28:
It could be that VoodooRMI isn't attaching then, especially if you have VoodooI2CHID as well. Probably worth checking `ioreg` or IoRegistryExplorer.
With regards to VoodooInput, you do have a third copy that comes with VoodooPS2. As long as that one is enabled, enabling the copy in VoodooI2C or VoodooRMI will cause a panic.

Thisisauser6443, 2024-05-28:
Checked IOReg a while ago, and the trackpad did show `Transport` as I2C, under `VoodooI2CHIDDevice`. 

Also, if I disable all the related entries to VoodooI2C, leaving just VoodooRMI and VoodooPS2 enabled, then the trackpad does not work, so I'd assume RMI is not being attached, even if I2CHID is disabled

1Revenger1, 2024-05-29:
Does Hid-rmi attach in Linux or do you see a Synaptics I2C HID device attach in windows? It could just be that your touchpad doesn't support the RMI4 protocol, in which case it sounds like a bug fix in VoodooI2CHID would be needed.

If you want, you could send the report descriptor shown on the VoodooI2CHID device in IoRegistryExplorer.

Thisisauser6443, 2024-05-30:
Can you be a little bit more specific on what the "Report descriptor" is?. I don't have Windows installed, right now, so IOReg is my only option, for the time being

1Revenger1, 2024-05-30:
The report descriptor is a big list of numbers used by VoodooI2CHID to figure out the touchpad's capabilities and what the reports look like when the touchpad sends them to the OS. VoodooI2CHID exposes it as a device property, so you probably can see it with `ioreg` as well. Not entirely sure how though, it might list it if you don't limit the width.

1Revenger1, 2024-05-30:
Yep, would you be able to get a text copy of that? Can just copy directly from ioregistryexplorer if you double click on the data.

Thisisauser6443, 2024-05-30:
```
<05 01 09 02 a1 01 85 02 09 01 a1 00 05 09 19 01 29 02 15 00 25 01 75 01 95 02 81 02 95 06 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02 81 06 c0 c0 05 01 09 02 a1 01 85 18 09 01 a1 00 05 09 19 01 29 03 46 00 00 15 00 25 01 75 01 95 03 81 02 95 05 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02 81 06 c0 c0 06 00 ff 09 02 a1 01 85 20 09 01 a1 00 09 03 15 00 26 ff 00 35 00 46 ff 00 75 08 95 05 81 02 c0 c0 05 0d 09 05 a1 01 85 03 05 0d 09 22 a1 02 15 00 25 01 09 47 09 42 95 02 75 01 81 02 95 01 75 03 25 05 09 51 81 02 75 01 95 03 81 03 05 01 15 00 26 f8 04 75 10 55 0e 65 11 09 30 35 00 46 24 04 95 01 81 02 46 30 02 26 a0 02 09 31 81 02 c0 05 0d 09 22 a1 02 15 00 25 01 09 47 09 42 95 02 75 01 81 02 95 01 75 03 25 05 09 51 81 02 75 01 95 03 81 03 05 01 15 00 26 f8 04 75 10 55 0e 65 11 09 30 35 00 46 24 04 95 01 81 02 46 30 02 26 a0 02 09 31 81 02 c0 05 0d 09 22 a1 02 15 00 25 01 09 47 09 42 95 02 75 01 81 02 95 01 75 03 25 05 09 51 81 02 75 01 95 03 81 03 05 01 15 00 26 f8 04 75 10 55 0e 65 11 09 30 35 00 46 24 04 95 01 81 02 46 30 02 26 a0 02 09 31 81 02 c0 05 0d 09 22 a1 02 15 00 25 01 09 47 09 42 95 02 75 01 81 02 95 01 75 03 25 05 09 51 81 02 75 01 95 03 81 03 05 01 15 00 26 f8 04 75 10 55 0e 65 11 09 30 35 00 46 24 04 95 01 81 02 46 30 02 26 a0 02 09 31 81 02 c0 05 0d 09 22 a1 02 15 00 25 01 09 47 09 42 95 02 75 01 81 02 95 01 75 03 25 05 09 51 81 02 75 01 95 03 81 03 05 01 15 00 26 f8 04 75 10 55 0e 65 11 09 30 35 00 46 24 04 95 01 81 02 46 30 02 26 a0 02 09 31 81 02 c0 05 0d 55 0c 66 01 10 47 ff ff 00 00 27 ff ff 00 00 75 10 95 01 09 56 81 02 09 54 25 7f 95 01 75 08 81 02 05 09 09 01 25 01 75 01 95 01 81 02 95 07 81 03 05 0d 85 08 09 55 09 59 75 04 95 02 25 0f b1 02 85 0d 09 60 75 01 95 01 15 00 25 01 b1 02 95 07 b1 03 85 07 06 00 ff 09 c5 15 00 26 ff 00 75 08 96 00 01 b1 02 c0 05 0d 09 0e a1 01 85 04 09 22 a1 02 09 52 15 00 25 0a 75 08 95 01 b1 02 c0 09 22 a1 00 85 06 09 57 09 58 75 01 95 02 25 01 b1 02 95 06 b1 03 c0 c0 06 00 ff 09 01 a1 01 85 09 09 02 15 00 26 ff 00 75 08 95 14 91 02 85 0a 09 03 15 00 26 ff 00 75 08 95 14 91 02 85 0b 09 04 15 00 26 ff 00 75 08 95 45 81 02 85 0c 09 05 15 00 26 ff 00 75 08 95 45 81 02 85 0f 09 06 15 00 26 ff 00 75 08 95 03 b1 02 85 0e 09 07 15 00 26 ff 00 75 08 95 01 b1 02 c0>
```

1Revenger1, 2024-05-30:
Interesting, looks like the trackpoint data does come through a HID report.
Give me a couple days, I think I can get you a version of VoodooI2CHID that works.
Going to close this for now since this uses VoodooI2CHID instead of VoodooRMI.

Thisisauser6443, 2024-05-30:
Grand, thanks for that

1Revenger1, 2024-06-01:
[VoodooI2C-2.8-RELEASE.zip](https://github.com/user-attachments/files/15520437/VoodooI2C-2.8-RELEASE.zip)
I've got no clue if this will work or not. Have a way to get back in if this causes a panic.

Thisisauser6443, 2024-06-01:
Yeah, didn’t work, unfortunately - Tested with both `I2CHID` and `Synaptics` kexts, and there was no kernel panic, but the trackpad, touchpoint and physical buttons were non-functioning after boot

1Revenger1, 2024-06-01:
Does an ioreg property show up showing the number of mice/track point devices it detected? It should show up under VoodooI2CPrecisionHIDEventDriver iirc.

Thisisauser6443, 2024-06-02:
The only entry I have that’s similar to that is `VoodooI2CPrecisionTouchpadHIDEventDriver`

1Revenger1, 2024-06-02:
Yep that would be it

Thisisauser6443, 2024-06-02:
[image]

1Revenger1, 2024-06-02:
[VoodooI2CHID.kext.zip](https://github.com/user-attachments/files/15525496/VoodooI2CHID.kext.zip)
Hrmm, I don't see what I am expecting there. Try this version - a property called "Trackpoints" should be there.

Thisisauser6443, 2024-06-03:
Yeah, no luck, unfortunately - I get a few `Invalid Parameters` with the new kext, and then the trackpad doesn't work after booting

Thisisauser6443, 2024-06-03:
I think I did, but I'll try again

1Revenger1, 2024-06-03:
Are you able to provide any log from OC? Should say what kext is causing the invalid parameter

Thisisauser6443, 2024-06-03:
OC says that the culprit is `VoodooI2CHID`

1Revenger1, 2024-06-03:
I think my local builds are completely broken, give me a day or two to figure this out.
Trying it locally is causing panics and other odd behavior which makes no sense.

Thisisauser6443, 2024-06-05:
Oof, alright

1Revenger1, 2024-06-11:
Sorry for dropping the ball on this. Once I figure out Sequoia stuff I'll get back to this. For now I've thrown my changes in here though: https://github.com/VoodooI2C/VoodooI2C/pull/551
Feel free to try and build it yourself.

1Revenger1, 2024-06-12:
[Archive.zip](https://github.com/user-attachments/files/15796878/Archive.zip)
Finally fixed the panic and loading - give this a shot and see if VoodooI2CPrecisionTouchpadHIDEventDriver finds anything.

Thisisauser6443, 2024-06-12:
[image]

The mouse buttons and trackpoint are recognised, by the looks of it - Only issue is that when I use either of them, the pointer goes straight to the bottom of the display, and if I keep using it, then it gradually makes its way to the lower-left corner

1Revenger1, 2024-06-12:
[VoodooI2CHID.kext.zip](https://github.com/user-attachments/files/15807695/VoodooI2CHID.kext.zip)
Want to give this a shot?

Thisisauser6443, 2024-06-12:
No dice - Same thing happens if I use the mouse buttons, just that the movement to the lower-left corner is a tad bit less erratic

1Revenger1, 2024-06-12:
[VI2C-CastUIntToSInt.zip](https://github.com/user-attachments/files/15808008/VI2C-CastUIntToSInt.zip)
Forgot to cast from an unsigned to a signed int, oops.

Thisisauser6443, 2024-06-12:
Lol, all good - It stopped the erratic movement, but any clicks from the buttons aren't being registered, and the trackpoint doesn't look like it's working either

1Revenger1, 2024-06-14:
[VI2C-Log.kext.zip](https://github.com/user-attachments/files/15841123/VI2C-Log.kext.zip)
This has some additional logging - mind giving this a spin? After using the touchpad, run `sudo dmesg | grep -i i2c` and it should show the delta x, y, and button values coming from the trackpoint. It may be a little easier to do this if you have `msgbuf=1048576` in your boot-args.

Thisisauser6443, 2024-06-14:
Uh, problem, the trackpad doesn’t work with the custom build. I still managed to get the output of the `dmesg` command, but I’m guessing it’s fairly useless in this case

```
[    0.693909]: VoodooI2CPCILakeController::pci8086,2e9 Starting I2C controller
[    0.693944]: VoodooI2CPCILakeController::pci8086,2e9 Set PCI power state D0
[    0.693948]: VoodooI2CPCILakeController::pci8086,2e9 Current CPU is Comet Lake or Ice Lake, patching...
[    0.693980]: VoodooI2CPCILakeController::pci8086,2e9 Publishing nub
[    0.694070]: VoodooI2CControllerDriver::pci8086,2e9 Probing controller
[    0.694086]: VoodooI2CControllerDriver::pci8086,2e9 Found valid Synopsys component, continuing with initialisation
[    0.694116]: VoodooI2CControllerNub::pci8086,2e9 SSCN not implemented in ACPI tables
[    0.694121]: VoodooI2CControllerNub::pci8086,2e9 FMCN not implemented in ACPI tables
[    0.694124]: VoodooI2CControllerDriver::pci8086,2e9 Warning: Error getting bus config, using defaults where necessary
[    0.694224]: VoodooI2CControllerDriver::pci8086,2e9 Publishing device nubs
[    0.695358]: VoodooI2CPCILakeController::pci8086,2e8 Starting I2C controller
[    0.695458]: VoodooI2CPCILakeController::pci8086,2e8 Set PCI power state D0
[    0.695464]: VoodooI2CPCILakeController::pci8086,2e8 Current CPU is Comet Lake or Ice Lake, patching...
[    0.695547]: VoodooI2CPCILakeController::pci8086,2e8 Publishing nub
[    0.696281]: VoodooI2CControllerDriver::pci8086,2e8 Probing controller
[    0.696351]: VoodooI2CControllerDriver::pci8086,2e8 Found valid Synopsys component, continuing with initialisation
[    0.696526]: VoodooI2CControllerNub::pci8086,2e8 SSCN not implemented in ACPI tables
[    0.696535]: VoodooI2CControllerNub::pci8086,2e8 FMCN not implemented in ACPI tables
[    0.696537]: VoodooI2CControllerDriver::pci8086,2e8 Warning: Error getting bus config, using defaults where necessary
[    0.696657]: VoodooI2CControllerDriver::pci8086,2e8 Publishing device nubs
[    0.696664]: VoodooI2CControllerDriver::pci8086,2e8 Found I2C device: SYNA30B3
[    0.697045]: VoodooI2CDeviceNub::TPD0 Found valid resources from _CRS method
[    0.697755]: VoodooI2CDeviceNub::TPD0 Found valid resources from _DSM or XDSM method
[    0.697768]: VoodooI2CDeviceNub::TPD0 Prefer resources from _DSM or XDSM method
[    0.697785]: VoodooI2CDeviceNub::TPD0 Found valid GPIO interrupts
[    0.697809]: VoodooI2CDeviceNub::TPD0 Got GPIO Controller! VoodooGPIOCannonLakeLP
[   11.147538]: u> 11147536 I2CAUXLOG: +dpAuxChWrite: MOTWR@0x00050 (0) - Status:0
[   11.147785]: u> 11147783 I2CAUXLOG: +dpAuxChWrite: MOTWR@0x00050 (1) - Status:0
[   11.147791]: WriteI2COverAUX:791 AuxData [0xfffffffddf82f64f:1, 32:1]
[   11.147923]: u> 11147921 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00050 (0) : Status: 0
[   11.148285]: u> 11148284 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00050 (10) : Status: 0
[   11.148290]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f920:16, 32:1]
[   11.148664]: u> 11148662 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00050 (10) : Status: 0
[   11.148669]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f930:16, 32:1]
[   11.148852]: 1 duplicate report for Sandbox: com.apple.AmbientDisplayAgent(336) deny(1) mach-lookup com.apple.windowserverSandbox: locationd(334) deny(1) mach-lookup com.apple.tccdu> 11149039 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00050 (10) : Status: 0
[   11.149045]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f940:16, 32:1]
[   11.149410]: Sandbox: locationd(334) deny(1) mach-lookup com.apple.tccd.systemu> 11149416 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00050 (10) : Status: 0
[   11.149422]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f950:16, 32:1]
[   11.149796]: u> 11149794 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00050 (10) : Status: 0
[   11.149801]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f960:16, 32:1]
[   11.149992]: Sandbox: systemstats(101) deny(1) mach-lookup com.apple.tccdu> 11150170 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00050 (10) : Status: 0
[   11.150177]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f970:16, 32:1]
[   11.150408]: Sandbox: systemstats(101) deny(1) mach-lookup com.apple.tccd.systemu> 11150552 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00050 (10) : Status: 0
[   11.150559]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f980:16, 32:1]
[   11.150642]: Sandbox: locationd(334) deny(1) mach-lookup com.apple.tccdSandbox: systemstats(101) deny(1) mach-lookup com.apple.tccdu> 11150929 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00050 (10) : Status: 0
[   11.150936]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f990:16, 32:1]
[   11.150993]: Sandbox: locationd(334) deny(1) mach-lookup com.apple.tccd.systemu> 11151068 I2CAUXLOG: +dpAuxChRead: I2CRD@0x00050 (0) : Status: 0
[   11.151203]: Sandbox: systemstats(101) deny(1) mach-lookup com.apple.tccd.systemu> 11151615 I2CAUXLOG: +dpAuxChWrite: MOTWR@0x00040 (0) - Status:0
[   11.152104]: u> 11152102 I2CAUXLOG: +dpAuxChWrite: MOTWR@0x00040 (1) - Status:0
[   11.152109]: WriteI2COverAUX:791 AuxData [0xfffffffddf82f528:1, 32:1]
[   11.152724]: u> 11152722 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00040 (10) : Status: 0
[   11.152730]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f640:16, 32:1]
[   11.153336]: u> 11153334 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00040 (10) : Status: 0
[   11.153341]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f650:16, 32:1]
[   11.153882]: Sandbox: systemstats(101) deny(1) mach-lookup com.apple.tccdu> 11153955 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00040 (10) : Status: 0
[... cut ...]
[   11.155829]: u> 11155826 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00040 (10) : Status: 0
[   11.155834]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f690:16, 32:1]
[   11.156445]: u> 11156443 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00040 (10) : Status: 0
[   11.156451]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f6a0:16, 32:1]
[   11.157065]: u> 11157063 I2CAUXLOG: +dpAuxChRead: MOTWR@0x00040 (10) : Status: 0
[   11.157071]: ReadI2COverAUX:738 AuxData [0xfffffffddf82f6b0:16, 32:1]
[   11.157556]: u> 11157554 I2CAUXLOG: +dpAuxChRead: I2CRD@0x00040 (0) : Status: 0
```

1Revenger1, 2024-06-14:
Weird, guessing that VoodooI2CHID didn't inject. Is this using the copy from when I fixed the [signed values](https://github.com/VoodooSMBus/VoodooRMI/issues/178#issuecomment-2163443693)?
Here is a new copy with both VoodooI2C and VoodooI2CHID:
[VI2C-LogV2.zip](https://github.com/user-attachments/files/15841523/VI2C-LogV2.zip)

Thisisauser6443, 2024-06-14:
Yeah, that’s solved it. There’s a lot more coming from VoodooI2CPrecisionTouchpadHIDEventDriver, and [from what I can see in the logs](https://github.com/user-attachments/files/15841630/Untitled.txt), it does seem to be displaying the dx/dy co-ordinates, just that a lot of the lines are duplicates of each other

1Revenger1, 2024-06-14:
[VI2C-LogNativeEngine.zip](https://github.com/user-attachments/files/15841853/VI2C-LogNativeEngine.zip)
Got another one to try out. That log looks like what I would expect.

Thisisauser6443, 2024-06-14:
Right, so the behaviour is a little different on this one - The movement of the pointer when I use the trackpoint is a a bit jumpy/stuttery/delayed, and the movement is a little bit random (as in, I can’t deduce any significant pattern of movement)

Here are the logs which I got using `dmesg` in case they’re of any use
[Untitled.txt](https://github.com/user-attachments/files/15842101/Untitled.txt)

1Revenger1, 2024-06-14:
I'm not entirely sure where to go from here as the log looks fairly normal. I got one last version where I divide the delta values by 5 but otherwise I got nothing else :(
[VI2C-DeltaDivisor.zip](https://github.com/user-attachments/files/15842308/VI2C-DeltaDivisor.zip)

Thisisauser6443, 2024-06-14:
Well, it’s bad news, again - The trackpoint and buttons aren’t registered at all

1Revenger1, 2024-06-14:
Any logs from VoodooI2CHID?

Thisisauser6443, 2024-06-16:
[Logs.txt](https://github.com/user-attachments/files/15859494/Logs.txt)

I should mention that the same hovering issue is back, from before

zhen-zen, 2024-06-20:
It's interesting that I didn't hit much issue back on an 850G7 where I added F17 support.

That touchpad is reported as SYNA30B4 but I guess it's less useful than VID/PID 06CB:CE09 and VersionID 0x2107.

```
            Method (_HID, 0, Serialized)  // _HID: Hardware ID
            {
                Switch (HPID)
                {
                    Case (0x103C00B0)
                    {
                        Return ("SYNA30B1")
                    }
                    Case (0x103C00B8)
                    {
                        Return ("SYNA30B1")
                    }
                    Case (0x103C00B2)
                    {
                        Return ("SYNA30B3")
                    }
                    Case (0x103C00BA)
                    {
                        Return ("SYNA30B3")
                    }
                    Case (0x103C00B5)
                    {
                        Return ("SYNA30B3")
                    }
                    Case (0x103C00BD)
                    {
                        Return ("SYNA30B3")
                    }
                    Case (0x103C00B3)
                    {
                        Return ("SYNA30B4")
                    }
                    Case (0x103C00BB)
                    {
                        Return ("SYNA30B4")
                    }
                    Case (0x103C00B7)
                    {
                        Return ("SYNA30B4")
                    }
                    Case (0x103C00BF)
                    {
                        Return ("SYNA30B4")
                    }
                    Default
                    {
                        Return ("SYNA30B1")
                    }

                }
            }
```

I forgot whether VoodooI2CHID has button support but RMII2C should work as expected.