VoodooSMBus/VoodooRMI #179: No support for SYNA329D

kuguazhiwang, 2024-06-07:
Computer: Lenove yoya 14s R7 5800H  
macOS version: macOS sonoma 14.4
Open Core version: (latest as of posting)

Device Manager:
[image]

Bug description:
When I use VoodooRMI (along with Voodoo I2C), in macOS recovery/installer/installed, it fails to work. The device manager hardware IDs are shown above. The kexts are all loaded successfully, checking the log.

I hope there are tutorials that can help us DIY or solve problems

1Revenger1, 2024-06-07:
Is this using the latest commit of VoodooI2C? The latest release of VoodooI2C does not include support for AMD I2C Controllers.

closed 2024-06-13 by 1Revenger1

kuguazhiwang, 2024-06-21:
https://github.com/VoodooI2C/VoodooI2C/pull/547#issuecomment-2076129321   
 and  add :NVRAM->boot-args  -vi2c-force-polling   
 can be useful ！