VoodooSMBus/VoodooRMI #184: VoodooRMI 1.4 doesnt work on my XPS15

gujiangjiang, 2024-10-07:
[image]

Here are the kext list
The VoodooPS2 VoodooI2C VoodooRMI VoodooInput are all latest version.

Using VoodooRMI 1.3.5 the trackpad works perfectly bug when updated to 1.4.0 the trackpad not works now.
My touchpad is synaptic I2C trackpad.

1Revenger1, 2024-10-08:
Are you able to provide logs? Would you be willing to go through [here](https://dortania.github.io/builds/?product=VoodooRMI&viewall=true) as well and bisect commits from [4d61d8b](https://github.com/VoodooSMBus/VoodooRMI/commit/4d61d8b3d2b372278be394b682ab0a306ca1295b) to [fb93f3f](https://github.com/VoodooSMBus/VoodooRMI/commit/fb93f3f1b1d4285fca5f9e5b5db6a55979ef5ef3)?

gujiangjiang, 2024-10-08:
> Are you able to provide logs? Would you be willing to go through [here](https://dortania.github.io/builds/?product=VoodooRMI&viewall=true) as well and bisect commits from [4d61d8b](https://github.com/VoodooSMBus/VoodooRMI/commit/4d61d8b3d2b372278be394b682ab0a306ca1295b) to [fb93f3f](https://github.com/VoodooSMBus/VoodooRMI/commit/fb93f3f1b1d4285fca5f9e5b5db6a55979ef5ef3)?您能提供日志吗？您是否愿意也浏览一下这里，并将 4d61d8b 到 fb93f3f 的提交一分为二？

I will try evey version from [4d61d8b](https://github.com/VoodooSMBus/VoodooRMI/commit/4d61d8b3d2b372278be394b682ab0a306ca1295b) to [fb93f3f](https://github.com/VoodooSMBus/VoodooRMI/commit/fb93f3f1b1d4285fca5f9e5b5db6a55979ef5ef3)

give me some time to get logs and feedback. thanks.

gujiangjiang, 2024-10-08:
gujiangjiang@gujiangjiangdeMacBook-Pro-4 ~ % sudo dmesg | grep VRMI
gujiangjiang@gujiangjiangdeMacBook-Pro-4 ~ % 

no logs found.

gujiangjiang, 2024-10-08:
@1Revenger1 
Here are the logs with VoodooPS2 2.3.5 + VoodooRMI 1.3.5 and works perfectly.

[    1.740905]: VRMI - Debug: RMII2C::TPDX probing
[    1.741777]: VRMI - Debug: RMII2C::TPDX Trying to set mode, attempt 0
[    1.742096]: VRMI - Debug: RMII2C::TPDX mode set
[    2.244926]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    2.244931]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    2.244935]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[    2.244938]: VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
[    2.244941]: VRMI - Debug: Config DisableWhileTypingTimeout loaded: 7d0 -> fa
[    2.244944]: VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 7d0 -> fa
[    2.244947]: VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
[    2.244950]: VRMI - Debug: Config ForceTouchType loaded: 1 -> 1
[    2.244952]: VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
[    2.244955]: VRMI - Debug: Config FingerMajorMinorDiffMax loaded: a -> a
[    2.244957]: VRMI - Debug: Config PalmRejectionWidth loaded: f -> a
[    2.244960]: VRMI - Debug: Config PalmRejectionHeight loaded: 50 -> 3c
[    2.244962]: VRMI - Debug: Config PalmRejectionTrackpointHeight loaded: 14 -> 14
[    2.244965]: VRMI - Debug: Updating Configuration
[    2.244974]: VRMI - Info: RMI Bus (DEBUG) Starting up!
[    2.246777]: VRMI - Debug: RMII2C::TPDX mode set
[    2.246796]: VRMI - Info: RMII2C::TPDX reset completed
[    2.246798]: VRMI - Error: Unable to reset
[    2.246800]: VRMI - Error: RMI initial reset failed! Continuing in spite of this
[    2.247404]: VRMI - Debug: rmi_probe_interrupts: Counting IRQs
[    2.267729]: VRMI - Debug: IRQ Count: 7
[    2.267806]: VRMI - Debug: rmi_init_functions: Creating functions
[    2.269554]: VRMI - Info: Initializing F34.
[    2.269558]: VRMI - Debug: Function F34 - IRQs: 1 CMD Base: 0 CTRL Base: 12 DATA Base: 0 QRY Base: 34 VER: 1
[    2.269562]: VRMI - Info: F34 not implemented
[    2.270268]: VRMI - Info: Initializing F01.
[    2.270271]: VRMI - Debug: Function F1 - IRQs: 1 CMD Base: 33 CTRL Base: 16 DATA Base: 4 QRY Base: 38 VER: 0
[    2.275590]: VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM3125-001, fw id: 1961136
[    2.278265]: VRMI - Info: Initializing F12.
[    2.278283]: VRMI - Debug: Function F12 - IRQs: 2 CMD Base: 0 CTRL Base: 22 DATA Base: 7 QRY Base: 67 VER: 0
[    2.281205]: VRMI - Debug: F12 - reg: 0 reg size: 1 subpackets: 1
[    2.281208]: VRMI - Debug: F12 - reg: 1 reg size: 1 subpackets: 1
[    2.281229]: VRMI - Debug: F12 - reg: 2 reg size: 3 subpackets: 2
[    2.281232]: VRMI - Debug: F12 - reg: 3 reg size: 31 subpackets: 12
[    2.281235]: VRMI - Debug: F12 - reg: 4 reg size: 1 subpackets: 1
[    2.281237]: VRMI - Debug: F12 - reg: 5 reg size: 5 subpackets: 2
[    2.281241]: VRMI - Debug: F12 - reg: 6 reg size: 25 subpackets: 11
[    2.281243]: VRMI - Debug: F12 - reg: 7 reg size: 1 subpackets: 1
[    2.281246]: VRMI - Debug: F12 - reg: 8 reg size: 3 subpackets: 2
[    2.281249]: VRMI - Debug: F12 - reg: 9 reg size: 10 subpackets: 5
[    2.281252]: VRMI - Debug: F12 - reg: 10 reg size: 1 subpackets: 1
[    2.281255]: VRMI - Debug: F12 - reg: 13 reg size: 1 subpackets: 1
[    2.283572]: VRMI - Debug: F12 - reg: 8 reg size: 10 subpackets: 3
[    2.283576]: VRMI - Debug: F12 - reg: 9 reg size: 15 subpackets: 3
[    2.283596]: VRMI - Debug: F12 - reg: 10 reg size: 17 subpackets: 3
[    2.283599]: VRMI - Debug: F12 - reg: 11 reg size: 15 subpackets: 4
[    2.283602]: VRMI - Debug: F12 - reg: 15 reg size: 7 subpackets: 3
[    2.283605]: VRMI - Debug: F12 - reg: 19 reg size: 67 subpackets: 5
[    2.283608]: VRMI - Debug: F12 - reg: 20 reg size: 1 subpackets: 1
[    2.283611]: VRMI - Debug: F12 - reg: 21 reg size: 2 subpackets: 1
[    2.283613]: VRMI - Debug: F12 - reg: 22 reg size: 1 subpackets: 1
[    2.283616]: VRMI - Debug: F12 - reg: 23 reg size: 2 subpackets: 2
[    2.283619]: VRMI - Debug: F12 - reg: 28 reg size: 1 subpackets: 1
[    2.285437]: VRMI - Debug: F12 - reg: 1 reg size: 40 subpackets: 5
[    2.285440]: VRMI - Debug: F12 - reg: 2 reg size: 1 subpackets: 1
[    2.285442]: VRMI - Debug: F12 - reg: 5 reg size: 15 subpackets: 5
[    2.285445]: VRMI - Debug: F12 - reg: 9 reg size: 10 subpackets: 5
[    2.285448]: VRMI - Debug: F12 - reg: 15 reg size: 1 subpackets: 1
[    2.285451]: VRMI - Debug: F12 - Data packet size: 0x43
[    2.286216]: VRMI - Debug: F12 - Number of fingers 5
[    2.287779]: VRMI - Info: Initializing F54.
[    2.287781]: VRMI - Debug: Function F54 - IRQs: 1 CMD Base: 301 CTRL Base: 269 DATA Base: 256 QRY Base: 302 VER: 0
[    2.287786]: VRMI - Info: F54 not implemented
[    2.289390]: VRMI - Info: Initializing F30.
[... 113 lines cut ...]
[   34.862941]: VRMI - Debug: F12 Packet
[   34.862964]: VRMI - Debug: Finger num: 0 (valid) (700, 350) [Z: 58 WX: 5 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[   34.870668]: VRMI - Debug: F12 Packet
[   34.870677]: VRMI - Debug: Finger num: 0 (valid) (702, 344) [Z: 58 WX: 5 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[   34.878543]: VRMI - Debug: F12 Packet
[   34.878552]: VRMI - Debug: Finger num: 0 (valid) (702, 341) [Z: 58 WX: 5 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[   34.886103]: VRMI - Debug: F12 Packet
[   34.886109]: VRMI - Debug: Finger num: 0 (valid) (702, 338) [Z: 58 WX: 5 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[   34.893937]: VRMI - Debug: F12 Packet
[... cut ...]
[   34.932909]: VRMI - Debug: F12 Packet
[   34.932917]: VRMI - Debug: Finger num: 0 (valid) (691, 327) [Z: 57 WX: 5 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[   34.940490]: VRMI - Debug: F12 Packet
[   34.940506]: VRMI - Debug: Finger num: 0 (valid) (688, 325) [Z: 55 WX: 5 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[   34.948244]: VRMI - Debug: F12 Packet
[   34.948251]: VRMI - Debug: Finger num: 0 (valid) (687, 324) [Z: 52 WX: 4 WY: 4 FingerType: 2 Pressure : 0 Button: 0]
[   34.955123]: VRMI - Debug: F12 Packet

1Revenger1, 2024-10-08:
Hrmm oh I see what might be the issue.
Give this a shot - I forgot to set a ready bit in the I2C driver.
[VoodooRMI-1.4.1-DEBUG.zip](https://github.com/user-attachments/files/17293667/VoodooRMI-1.4.1-DEBUG.zip)

PR #185 (2024-10-08): Fix ready bit not being set on startup
Fixes #184
closed 2024-10-08 by gujiangjiang