VoodooSMBus/VoodooRMI #186: draging with look and without look not work only work 3finger darg

Edwardwich, 2024-10-08:
draging with look and without look not work only work 3finger darg :

VoodooRMI version : 1.4.0
VoodooPS2 version :2.3.6.

[image]

[image]

sudo dmesg:
[log.txt](https://github.com/user-attachments/files/17295871/log.txt)

iotrg:
[Ali’s MacBook Pro.ioreg.zip](https://github.com/user-attachments/files/17295876/Ali.s.MacBook.Pro.ioreg.zip)

1Revenger1, 2024-10-08:
How is Drag with or without lock supposed to work? Appears to work on my side on Sonoma - I can double tap the top of the window then start moving it around. It'll stay locked to the window depending on the drag lock setting.

Edwardwich, 2024-10-08:
> How is Drag with or without lock supposed to work? Appears to work on my side on Sonoma - I can double tap the top of the window then start moving it around. It'll stay locked to the window depending on the drag lock setting.

yes in Sonoma work but in sequoia not working

Edwardwich, 2024-10-08:
and also drag with look and without look work in mac os sequoia when useing VoodooRMI version : 1.3.5

1Revenger1, 2024-10-12:
I do not have a laptop running Sequoia unfortunately, so am unable to reproduce this. Just as a quick sanity check, is tap to click enabled under the Sequoia touchpad settings? I think it is enabled automatically with drag lock but it is probably worth double checking.

Some tests I guess - Can you try just replacing VoodooSMBus with the older version while keeping the newer VoodooRMI?
Second set of sets is trying out a few different commits of VoodooRMI. The ones that make changes to input handling between 1.3.5 and 1.4 is
fb93f3f1b1d4285fca5f9e5b5db6a55979ef5ef3
3f0dd535cb3f347024939dd4f54be4e18a5fb978
e73d53371d97b6c04ae57e2fe73bc10a6010b45c
341ff046ca11aa310187a0dcbf06e047f18e8fae

The different builds of VoodooRMI can be grabbed here: https://dortania.github.io/builds/?product=VoodooRMI&viewall=true

Edwardwich, 2024-10-12:
[image]
[image]

I try older verions of  VoodooSMBus with last VoodooRMI and tell u

Edwardwich, 2024-10-12:
i try voodoo rmi 1.4.1 with VoodooSMBus 1.3.5 but all gestures is broken 
[image]
and also tap to clik not worl

Edwardwich, 2024-10-12:
[image]

Edwardwich, 2024-10-12:
[Ali’s MacBook Pro.ioreg.zip](https://github.com/user-attachments/files/17349542/Ali.s.MacBook.Pro.ioreg.zip)

Edwardwich, 2024-10-12:
> I do not have a laptop running Sequoia unfortunately, so am unable to reproduce this. Just as a quick sanity check, is tap to click enabled under the Sequoia touchpad settings? I think it is enabled automatically with drag lock but it is probably worth double checking.
> 
> Some tests I guess - Can you try just replacing VoodooSMBus with the older version while keeping the newer VoodooRMI? Second set of sets is trying out a few different commits of VoodooRMI. The ones that make changes to input handling between 1.3.5 and 1.4 is [fb93f3f](https://github.com/VoodooSMBus/VoodooRMI/commit/fb93f3f1b1d4285fca5f9e5b5db6a55979ef5ef3) [3f0dd53](https://github.com/VoodooSMBus/VoodooRMI/commit/3f0dd535cb3f347024939dd4f54be4e18a5fb978) [e73d533](https://github.com/VoodooSMBus/VoodooRMI/commit/e73d53371d97b6c04ae57e2fe73bc10a6010b45c) [341ff04](https://github.com/VoodooSMBus/VoodooRMI/commit/341ff046ca11aa310187a0dcbf06e047f18e8fae)
> 
> The different builds of VoodooRMI can be grabbed here: https://dortania.github.io/builds/?product=VoodooRMI&viewall=true

hi I fork ur repo and revert all code in fb93f3f and now every thing work great with last voodoo ps2 kext 

my voodoo rmi repo : https://github.com/Edwardwich/VoodooRMI/tree/fb93f3f

1Revenger1, 2024-10-12:
[VoodooRMI-1.4.2-DEBUG.zip](https://github.com/user-attachments/files/17350781/VoodooRMI-1.4.2-DEBUG.zip)
Mind giving this a shot?

Edwardwich, 2024-10-12:
> [VoodooRMI-1.4.2-DEBUG.zip](https://github.com/user-attachments/files/17350781/VoodooRMI-1.4.2-DEBUG.zip) Mind giving this a shot?

everything work grate now thank you 
[Ali’s MacBook Pro.ioreg.zip](https://github.com/user-attachments/files/17350898/Ali.s.MacBook.Pro.ioreg.zip)
[Terminal Saved Output.txt](https://github.com/user-attachments/files/17350905/Terminal.Saved.Output.txt)

1Revenger1, 2024-10-12:
Thanks! I've gone ahead and pushed the changes to master.

closed 2024-10-12 by 1Revenger1

Edwardwich, 2024-10-12:
> Thanks! I've gone ahead and pushed the changes to master.

already I download and use it TNX a lot

PR #191 (2024-10-12): Set contact count to max number of fingers
Fixes #186 

If a finger is lifted (or tapped in quick succession), the current finger is not guaranteed to be in the first slot