VoodooSMBus/VoodooRMI #188: Panic after updating to version 1.4.1

HeartAndSoul001, 2024-10-09:
**Describe the bug**
Panic after updating to version 1.4.1, the screen will be stuck and can not perform any operation

**To Reproduce**
panic may be caused when something about system hibernation append(sleep or wake up)

**Expected behavior**
screen stuck, fan turning wildly

**Log**
	{"bug_type":"210","timestamp":"2024-10-09 15:17:28.00 +0800","os_version":"macOS 13.7 (22H123)","roots_installed":0,"incident_id":"20CA682B-4EED-4861-A6D3-C0EAE383E1BC"}
{"macOSProcessedStackshotData":"bm8gb24gZGlzayBwYW5pYyBzdGFja3Nob3QgZm91bmQgaW4gY29yZWZpbGU=","macOSPanicString":"panic(cpu 0 caller 0xffffff800e9b5293): Kernel trap at 0xffffff8000000001, type 13=general protection, registers:\nCR0: 0x000000008001003b, CR2: 0x0000600000f0c2a0, CR3: 0x000000001b884000, CR4: 0x00000000003626e0\nRAX: 0x0000000000000000, RBX: 0x0000b1184e0000c1, RCX: 0x0000000000000000, RDX: 0x0000000001000000\nRSP: 0xffffffbab7487dc0, RBP: 0x0000000000000000, RSI: 0xffffffaa6806ced0, RDI: 0x0000000000000000\nR8:  0x0000000000000002, R9:  0x0000000000000000, R10: 0xffffff800f5221b8, R11: 0x0000000000000003\nR12: 0xffffff800f51c1c0, R13: 0xffffffa59b143620, R14: 0x0000000000000000, R15: 0xffffffaa67653b80\nRFL: 0x0000000000010286, RIP: 0xffffff8000000001, CS:  0x0000000000000008, SS:  0x0000000000000010\nFault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 0\n\nPanicked task 0xffffffa0cec64698: 174 threads: pid 0: kernel_task\nBacktrace (CPU 0), panicked thread: 0xffffff9c0225bb30, Frame : Return Address\n0xffffff800e50c1a0 : 0xffffff800e870c7d \n0xffffff800e50c1f0 : 0xffffff800e9c52e4 \n0xffffff800e50c230 : 0xffffff800e9b4df7 \n0xffffff800e50c280 : 0xffffff800e811971 \n0xffffff800e50c2a0 : 0xffffff800e870f5d \n0xffffff800e50c390 : 0xffffff800e870607 \n0xffffff800e50c3f0 : 0xffffff800efdb8db \n0xffffff800e50c4e0 : 0xffffff800e9b5293 \n0xffffff800e50c660 : 0xffffff800e9b4f7d \n0xffffff800e50c6b0 : 0xffffff800e811971 \n0xffffff800e50c6d0 : 0xffffff8000000001 \n\nProcess name corresponding to current thread (0xffffff9c0225bb30): kernel_task\nBoot args: brcmfx-country=CN\n\nMac OS version:\n22H123\n\nKernel version:\nDarwin Kernel Version 22.6.0: Wed Jul 31 21:42:48 PDT 2024; root:xnu-8796.141.3.707.4~1\/RELEASE_X86_64\nKernel UUID: 15D2842B-072B-3795-BE93-B063FF3D4AA0\nroots installed: 0\nKernelCache slide: 0x000000000e400000\nKernelCache base:  0xffffff800e600000\nKernel slide:      0x000000000e4dc000\nKernel text base:  0xffffff800e6dc000\n__HIB  text base: 0xffffff800e500000\nSystem model name: MacBookPro14,2 (Mac-CAD6701F7CEA0921)\nSystem shutdown begun: YES\nPanic diags file available: YES (0x0)\nHibernation exit count: 0\n\nSystem uptime in nanoseconds: 306221475972\nLast Sleep:           absolute           base_tsc          base_nano\n  Uptime  : 0x000000474c38eca9\n  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000\n  Wake    : 0x0000000000000000 0x0000001291863c82 0x0000000000000000\nCompressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space\nZone info:\n  Zone map: 0xffffff90cdcad000 - 0xffffffb0cdcad000\n  . PGZ   : 0xffffff90cdcad000 - 0xffffff90cfcae000\n  . VM    : 0xffffff90cfcae000 - 0xffffff959c4ad000\n  . RO    : 0xffffff959c4ad000 - 0xffffff9735cad000\n  . GEN0  : 0xffffff9735cad000 - 0xffffff9c024ad000\n  . GEN1  : 0xffffff9c024ad000 - 0xffffffa0cecad000\n  . GEN2  : 0xffffffa0cecad000 - 0xffffffa59b4ad000\n  . GEN3  : 0xffffffa59b4ad000 - 0xffffffaa67cad000\n  . DATA  : 0xffffffaa67cad000 - 0xffffffb0cdcad000\n  Metadata: 0xffffff8053ad9000 - 0xffffff8073ad9000\n  Bitmaps : 0xffffff8073ad9000 - 0xffffff8076ad9000\n  Extra   : 0 - 0\n\nlast started kext at 76731653211: @filesystems.msdosfs\t1.10 (addr 0xffffff7fa49a5000, size 57344)\nlast stopped kext at 230799538406: >!AFIVRDriver\t4.1.0 (addr 0xffffff7fa3469000, size 8192)\nloaded kexts:\nme.kishorprins.VoodooInput\t1.1.6\ncom.1Revenger1.RMISMBus\t1.0\ncom.1Revenger1.VoodooRMI\t1.4.1\nas.acidanthera.voodoo.driver.PS2Mouse\t2.3.6\nas.acidanthera.voodoo.driver.PS2Trackpad\t2.3.6\nas.acidanthera.voodoo.driver.PS2Keyboard\t2.3.6\nde.leo-labs.VoodooSMBus\t3.0\nas.acidanthera.mieze.!IMausi\t1.0.7\nru.joedm.SMCSuperIO\t1.3.4\nas.acidanthera.voodoo.driver.PS2!C\t2.3.6\nas.vit9696.SMCProcessor\t1.3.4\nru.usrsse2.SMCLightSensor\t1.3.4\norg.acidanthera.NVMeFix\t1.1.1\nas.acidanthera.BrightnessKeys\t1.0.3\norg.acidanthera.driver.CPUFriend\t1.2.9\nas.vit9696.!AALC\t1.9.2\nas.lvs1974.AirportBrcmFixup\t2.1.9\ncom.1Revenger1.ECEnabler\t1.0.5\nas.vit9696.WhateverGreen\t1.6.8\nru.usrsse2.SMCBatteryManager\t1.3.4\nas.vit9696.VirtualSMC\t1.3.4\nas.vit9696.Lilu\t1.6.9\n@filesystems.msdosfs\t1.10\n>AudioAUUC\t1.70\n>X86PlatformShim\t1.0.0\n>AGPM\t131\n>!APlatformEnabler\t2.7.0d0\n@filesystems.autofs\t3.0\n>!AHDA\t440.2\n>!AGraphicsDevicePolicy\t7.1.19\n@AGDCPluginDisplayMetrics\t7.1.19\n>eficheck\t1\n>pmtelemetry\t1\n@filesystems.nfs\t1\n|IOUserEthernet\t1.0.1\n>usb.!UUserHCI\t1\n>!ABacklight\t180.8\n>!AUpstreamUserClient\t3.6.9\n>!AMCCSControl\t1.16\n>!ASMCLMU\t212\n>!AHV\t1\n>!ADiskImages2\t198.100.13\n>!A!ISlowAdaptiveClocking\t4.0.0\n>!A!IKBLGraphics\t20.0.7\n>!A!IPCHPMC\t2.0.1\n>!A!IKBLGraphicsFramebuffer\t20.0.7\n>!ATopCaseHIDEventDriver\t6440.7\n>!AFileSystemDriver\t3.0.1\n@filesystems.tmpfs\t1\n@filesystems.lifs\t1\n@filesystems.hfs.kext\t627.100.6\n@BootCache\t40\n@!AFSCompression.!AFSCompressionTypeZlib\t1.0.0\n@!AFSCompression.!AFSCompressionTypeDataless\t1.0.0d1\n>!ASmartBatteryManager\t161.0.0\n@filesystems.apfs\t2142.140.9\n>AirPort.BrcmNIC\t1400.1.1\n@private.KextAudit\t1.0\n>!AHPET\t1.8\n>!ARTC\t2.0.1\n>!AACPIButtons\t6.1\n>!ASMBIOS\t2.1\n>!AACPIEC\t6.1\n>!AAPIC\t1.7\n$!AUserConsent\t1\n@!ASystemPolicy\t2.0.0\n@nke.applicationfirewall\t404\n|IOKitRegistryCompatibility\t1\n|EndpointSecurity\t1\n@Dont_Steal_Mac_OS_X\t7.0.0\n@kec.Compression\t1\n@kec.!AEncryptedArchive\t1\n>DspFuncLib\t440.2\n@kext.OSvKernDSPLib\t530\n>!AGraphicsControl\t7.1.19\n>!AHDA!C\t440.2\n|IOHDA!F\t440.2\n@kext.triggers\t1.0\n>IOHIDPowerSource\t1\n>!ACallbackPowerSource\t1\n>!ABacklightExpert\t1.1.0\n|IOAudio!F\t440.2\n@vecLib.kext\t1.2.0\n>!AAudioClockLibs\t240.1\n>!ASMBus!C\t1.0.18d1\n>X86PlatformPlugin\t1.0.0\n|IO!BSerialManager\t9.0.0\n|IO!BPacketLogger\t9.0.0\n|IO!BHost!CUSBTransport\t9.0.0\n|IO!BHost!CUARTTransport\t9.0.0\n|IO!BHost!CTransport\t9.0.0\n>IO!BHost!CPCIeTransport\t9.0.0\n|IOAVB!F\t1140.6\n@plugin.IOgPTPPlugin\t1150.2\n|CSR!BHost!CUSBTransport\t9.0.0\n|Broadcom!BHost!CUSBTransport\t9.0.0\n|Broadcom!B20703USBTransport\t9.0.0\n>!ARSMChannel\t1\n|IORSM!F\t1\n>!AIPAppender\t1.0\n@!AGPUWrangler\t7.1.19\n|IOSlowAdaptiveClocking!F\t1.0.0\n>IOPlatformPlugin!F\t6.0.0d8\n|IONDRVSupport\t597\n|IOAccelerator!F2\t475.40.6\n@!AGraphicsDeviceControl\t7.1.19\n|IOGraphics!F\t597\n>driverkit.serial\t6.0.0\n|IOSerial!F\t11\n>!AHS!BDriver\t6440.7\n>IO!BHIDDriver\t9.0.0\n>!AActuatorDriver\t6440.7\n>!AMultitouchDriver\t6440.7\n>!AInputDeviceSupport\t6460.1\n>usb.IOUSBHostHIDDevice\t1.2\n>usb.cdc\t5.0.0\n>usb.networking\t5.0.0\n>usb.!UHostCompositeDevice\t1.2\n>usb.!UHub\t1.2\n>!UMergeNub\t900.4.2\n>!ABSDKextStarter\t3\n|IOSurface\t336.60.1\n@filesystems.hfs.encodings.kext\t1\n>!ASyntheticGame!C\t10.7.1\n>usb.!UHostPacketFilter\t1.0\n|IOUSB!F\t900.4.2\n>!AXsanScheme\t3\n|IO80211!FLegacy\t1200.12.2b1\n>corecapture\t1.0.4\n|IONVMe!F\t2.1.0\n>!A!ILpssGspi\t3.0.60\n>usb.!UXHCIPCI\t1.2\n>usb.!UXHCI\t1.2\n>!AEFINVRAM\t2.1\n>!AEFIRuntime\t2.1\n|IOHID!F\t2.0.0\n|IOSMBus!F\t1.1\n|IOTimeSync!F\t1150.2\n|IOSkywalk!F\t1.0\n>mDNSOffloadUserClient\t1.0.1b8\n|IONetworking!F\t3.4\n>DiskImages\t493.0.0\n|IO!B!F\t9.0.0\n|IOReport!F\t47\n$quarantine\t4\n$sandbox\t300.0\n@kext.!AMatch\t1.0.0d1\n>!ASSE\t1.0\n>!AKeyStore\t2\n>!UTDM\t554\n|IOUSBMass!SDriver\t235.140.2\n|IOSCSIBlockCommandsDevice\t482.120.2\n|IO!S!F\t2.1\n|IOSCSIArchitectureModel!F\t482.120.2\n>!AFDEKeyStore\t28.30\n>!AEffaceable!S\t1.0\n>!ACyrus\t1\n>!AMobileFileIntegrity\t1.0.5\n$!AImage4\t5.0.0\n@kext.CoreTrust\t1\n>!ACredentialManager\t1.0\n|CoreAnalytics!F\t1\n>KernelRelayHost\t1\n|IOUSBHost!F\t1.2\n>!UHostMergeProperties\t1.2\n>usb.!UCommon\t1.0\n>!ABusPower!C\t1.0\n>!ASEPManager\t1.0.1\n>IOSlaveProcessor\t1\n>!AACPIPlatform\t6.1\n>!ASMC\t3.1.9\n|IOPCI!F\t2.9\n|IOACPI!F\t1.4\n>watchdog\t1\n@kec.pthread\t1\n@kec.Libm\t1\n@kec.corecrypto\t12.0\n\n\n"}

1Revenger1, 2024-10-09:
Can you add `keepsyms=1` to your boot arguments and get a new panic dump? Looks like the panics don't show where the kexts are in memory without it so it is impossible for me to figure out where the panic is.

HeartAndSoul001, 2024-10-09:
@1Revenger1 Yes, off course I can, I will add boot arguments and get a new panic dump tomorrow because of the separation with my laptop. Thanks a lot for your reply and your wonderful works about VoodooRMI!

HeartAndSoul001, 2024-10-10:
@1Revenger1 hi, this is new panic dump with boot arguments keepsyms=1. To reproduce the panic, I update the VoodooRMI to version 1.4.1, and reboot to make new kext work, the panic will happen when I let the laptop hibernate. when panic, my laptop will automatically reboot and pop up a window where there is panic report. And, if I let the laptop reboot manually, the screen will be stuck and fan turn wildly, just like there is a dead cycle in computer process. Please help me with this panic and feel free to call me.
{"bug_type":"210","timestamp":"2024-10-10 08:02:48.00 +0800","os_version":"macOS 13.7 (22H123)","roots_installed":0,"incident_id":"606AD0DF-36F1-4ED7-A6E2-D340CF99326B"}
{"macOSProcessedStackshotData":"bm8gb24gZGlzayBwYW5pYyBzdGFja3Nob3QgZm91bmQgaW4gY29yZWZpbGU=","macOSPanicString":"panic(cpu 0 caller 0xffffff80075b5293): Kernel trap at 0xffffff8000000001, type 14=page fault, registers:\nCR0: 0x000000008001003b, CR2: 0x00006f1e200000f8, CR3: 0x0000000014484000, CR4: 0x00000000003626e0\nRAX: 0x0000000000000000, RBX: 0x00006f1e200000f8, RCX: 0x0000000000000000, RDX: 0x0000000001000000\nRSP: 0xfffffffe007a3dc0, RBP: 0x0000000000000000, RSI: 0xffffffbaa51f1600, RDI: 0x0000000000000000\nR8:  0x0000000000000002, R9:  0x0000000000000000, R10: 0xffffff80081221b8, R11: 0x0000000000000000\nR12: 0xffffff800811c1c0, R13: 0xffffffa10d5df700, R14: 0x76405e0000f93574, R15: 0xffffffaaa685cb00\nRFL: 0x0000000000010286, RIP: 0xffffff8000000001, CS:  0x0000000000000008, SS:  0x0000000000000000\nFault CR2: 0x00006f1e200000f8, Error code: 0x0000000000000002, Fault CPU: 0x0, PL: 0, VF: 1\n\nPanicked task 0xffffffaaa64698b8: 189 threads: pid 0: kernel_task\nBacktrace (CPU 0), panicked thread: 0xffffffa5d8ef2598, Frame : Return Address\n0xfffffffe007a37a0 : 0xffffff8007470c7d mach_kernel : _handle_debugger_trap + 0x4ad\n0xfffffffe007a37f0 : 0xffffff80075c52e4 mach_kernel : _kdp_i386_trap + 0x114\n0xfffffffe007a3830 : 0xffffff80075b4df7 mach_kernel : _kernel_trap + 0x3b7\n0xfffffffe007a3880 : 0xffffff8007411971 mach_kernel : _return_from_trap + 0xc1\n0xfffffffe007a38a0 : 0xffffff8007470f5d mach_kernel : _DebuggerTrapWithState + 0x5d\n0xfffffffe007a3990 : 0xffffff8007470607 mach_kernel : _panic_trap_to_debugger + 0x1a7\n0xfffffffe007a39f0 : 0xffffff8007bdb8db mach_kernel : _panic + 0x84\n0xfffffffe007a3ae0 : 0xffffff80075b5293 mach_kernel : _sync_iss_to_iks + 0x2c3\n0xfffffffe007a3c60 : 0xffffff80075b4f7d mach_kernel : _kernel_trap + 0x53d\n0xfffffffe007a3cb0 : 0xffffff8007411971 mach_kernel : _return_from_trap + 0xc1\n0xfffffffe007a3cd0 : 0xffffff8000000001 \n\nProcess name corresponding to current thread (0xffffffa5d8ef2598): kernel_task\nBoot args: brcmfx-country=CN keepsyms=1\n\nMac OS version:\n22H123\n\nKernel version:\nDarwin Kernel Version 22.6.0: Wed Jul 31 21:42:48 PDT 2024; root:xnu-8796.141.3.707.4~1\/RELEASE_X86_64\nKernel UUID: 15D2842B-072B-3795-BE93-B063FF3D4AA0\nroots installed: 0\nKernelCache slide: 0x0000000007000000\nKernelCache base:  0xffffff8007200000\nKernel slide:      0x00000000070dc000\nKernel text base:  0xffffff80072dc000\n__HIB  text base: 0xffffff8007100000\nSystem model name: MacBookPro14,2 (Mac-CAD6701F7CEA0921)\nSystem shutdown begun: NO\nPanic diags file available: YES (0x0)\nHibernation exit count: 0\n\nSystem uptime in nanoseconds: 202040631492\nLast Sleep:           absolute           base_tsc          base_nano\n  Uptime  : 0x0000002f0a8f648d\n  Sleep   : 0x0000000000000000 0x0000000000000000 0x0000000000000000\n  Wake    : 0x0000000000000000 0x0000000964477e98 0x0000000000000000\nCompressor Info: 0% of compressed pages limit (OK) and 0% of segments limit (OK) with 0 swapfiles and OK swap space\nZone info:\n  Zone map: 0xffffff9aa5452000 - 0xffffffbaa5452000\n  . PGZ   : 0xffffff9aa5452000 - 0xffffff9aa7453000\n  . VM    : 0xffffff9aa7453000 - 0xffffff9f73c52000\n  . RO    : 0xffffff9f73c52000 - 0xffffffa10d452000\n  . GEN0  : 0xffffffa10d452000 - 0xffffffa5d9c52000\n  . GEN1  : 0xffffffa5d9c52000 - 0xffffffaaa6452000\n  . GEN2  : 0xffffffaaa6452000 - 0xffffffaf72c52000\n  . GEN3  : 0xffffffaf72c52000 - 0xffffffb43f452000\n  . DATA  : 0xffffffb43f452000 - 0xffffffbaa5452000\n  Metadata: 0xfffffff3f32cc000 - 0xfffffff4132cc000\n  Bitmaps : 0xfffffff4132cc000 - 0xfffffff4162cc000\n  Extra   : 0 - 0\n\nlast started kext at 10757594809: @filesystems.autofs\t3.0 (addr 0xffffff7f9d56b000, size 32768)\nloaded kexts:\nme.kishorprins.VoodooInput\t1.1.6\ncom.1Revenger1.RMISMBus\t1.0\ncom.1Revenger1.VoodooRMI\t1.4.1\nas.acidanthera.voodoo.driver.PS2Mouse\t2.3.6\nas.acidanthera.voodoo.driver.PS2Trackpad\t2.3.6\nas.acidanthera.voodoo.driver.PS2Keyboard\t2.3.6\nru.joedm.SMCSuperIO\t1.3.4\nde.leo-labs.VoodooSMBus\t3.0\nas.acidanthera.mieze.!IMausi\t1.0.7\nas.acidanthera.voodoo.driver.PS2!C\t2.3.6\nas.vit9696.SMCProcessor\t1.3.4\nru.usrsse2.SMCLightSensor\t1.3.4\norg.acidanthera.NVMeFix\t1.1.1\nas.acidanthera.BrightnessKeys\t1.0.3\norg.acidanthera.driver.CPUFriend\t1.2.9\nas.vit9696.!AALC\t1.9.2\nas.lvs1974.AirportBrcmFixup\t2.1.9\ncom.1Revenger1.ECEnabler\t1.0.5\nas.vit9696.WhateverGreen\t1.6.8\nru.usrsse2.SMCBatteryManager\t1.3.4\nas.vit9696.VirtualSMC\t1.3.4\nas.vit9696.Lilu\t1.6.9\n@filesystems.autofs\t3.0\n>AudioAUUC\t1.70\n>X86PlatformShim\t1.0.0\n>AGPM\t131\n>!APlatformEnabler\t2.7.0d0\n>!AGraphicsDevicePolicy\t7.1.19\n@AGDCPluginDisplayMetrics\t7.1.19\n>pmtelemetry\t1\n@filesystems.nfs\t1\n|IOUserEthernet\t1.0.1\n>usb.!UUserHCI\t1\n>!AHDAHardwareConfigDriver\t440.2\n>!AUpstreamUserC [... cut ...] chitectureModel!F\t482.120.2\n>!AFDEKeyStore\t28.30\n>!AEffaceable!S\t1.0\n>!ACyrus\t1\n>!AMobileFileIntegrity\t1.0.5\n$!AImage4\t5.0.0\n@kext.CoreTrust\t1\n>!ACredentialManager\t1.0\n|CoreAnalytics!F\t1\n>KernelRelayHost\t1\n|IOUSBHost!F\t1.2\n>!UHostMergeProperties\t1.2\n>usb.!UCommon\t1.0\n>!ABusPower!C\t1.0\n>!ASEPManager\t1.0.1\n>IOSlaveProcessor\t1\n>!AACPIPlatform\t6.1\n>!ASMC\t3.1.9\n|IOPCI!F\t2.9\n|IOACPI!F\t1.4\n>watchdog\t1\n@kec.pthread\t1\n@kec.Libm\t1\n@kec.corecrypto\t12.0\n\n\n"}

HeartAndSoul001, 2024-10-10:
@1Revenger1 In addition, VoodooRMI Version 1.4.0 also make my system reboot as soon as the system go to sleep and cause a panic. But version 1.3.9 will not. Below is my SSDT list, and the SBUS ssdt file. I don't know if update to version >= 1.4.0, there are some changes about the config.plist or other config.

[image]

[image]

HeartAndSoul001, 2024-10-10:
[image]
[image]

1Revenger1, 2024-10-12:
My first thought is that the `BUS0` device isn't going to do anything while VoodooSMBus/VoodooRMI are being used. Maybe try removing that and just leaving the RCFG properties? Unfortunately I can see why the panic trace didn't give you addresses for VoodooRMI/SMBus - they aren't part of the stacktrace. Something must be getting clobbered but am unsure where. Would you be able to get me boot logs using dmesg? `msgbuf=1048576` in the boot-args as well as DebugEnhancer.kext should work. Once rebooted, run `sudo dmesg | grep -i vrmi > ~/Desktop/vrmi_log.txt` 

On an unrelated note, you may want to try out YogaSMC if you have a Thinkpad. It'll reset the LEDs on wakeup and give functionality to a lot of the function keys.

HeartAndSoul001, 2024-10-12:
@1Revenger1 yes, off course, here is the log:
[vrmi_log_01.txt](https://github.com/user-attachments/files/17348593/vrmi_log_01.txt)
[vrmi_log.txt](https://github.com/user-attachments/files/17348594/vrmi_log.txt)

1Revenger1, 2024-10-12:
Woah that is a weird log. VoodooRMI is showing up as booting up twice. Are you able to provide an ioreg from IORegistryExplorer real quick?

HeartAndSoul001, 2024-10-12:
here is the ioreg file:
[MacBook Pro.zip](https://github.com/user-attachments/files/17348625/MacBook.Pro.zip)

HeartAndSoul001, 2024-10-12:
Please, if you have any other questions, please feel free to contact me. Thanks a lot

1Revenger1, 2024-10-12:
Hrmm, looks like the root cause is that there are two copies of ApplePS2SMBusDevice appearing. Can you run `sudo dmesg | grep -iE 'ps2|vrmi' > ~/Desktop/ps2.txt`?

I think this may partially solve it, but I would like to continue debugging why two copies of ApplePS2SMBusDevice are appearing.
[VoodooSMBus.kext.zip](https://github.com/user-attachments/files/17348644/VoodooSMBus.kext.zip)

HeartAndSoul001, 2024-10-12:
here is the file:
[ps2.txt](https://github.com/user-attachments/files/17348652/ps2.txt)

1Revenger1, 2024-10-12:
Hrmm, quick question - did you re-add address 0x2C to VoodooSMBus' Info.plist?

HeartAndSoul001, 2024-10-12:
[image]
I have never modified the VoodooSMBus.kext. just let it be its original content from GitHub

HeartAndSoul001, 2024-10-12:
> Hrmm, looks like the root cause is that there are two copies of ApplePS2SMBusDevice appearing. Can you run `sudo dmesg | grep -iE 'ps2|vrmi' > ~/Desktop/ps2.txt`?
>
> I think this may partially solve [...]

with this, is looks like the kext is not be loaded, here is the log:
[ps2.txt](https://github.com/user-attachments/files/17348702/ps2.txt)
[vrmi_log.txt](https://github.com/user-attachments/files/17348703/vrmi_log.txt)
[MacBook Pro.zip](https://github.com/user-attachments/files/17348704/MacBook.Pro.zip)

1Revenger1, 2024-10-12:
Is VoodooPS2 modified at all? It looks like probing on the PS/2 side is happening twice and I am really confused how. Mind listing the kexts you have injected and upload your ACPI?

HeartAndSoul001, 2024-10-12:
no, there is no modification about VoodooPS2. here is my OC files and kext list:
[image]
[image]
[OC.zip](https://github.com/user-attachments/files/17348758/OC.zip)

1Revenger1, 2024-10-12:
[VoodooPS2Controller-2.3.6-RELEASE.zip](https://github.com/user-attachments/files/17348789/VoodooPS2Controller-2.3.6-RELEASE.zip)
Mind using this copy of VoodooPS2 and uploading a new PS2 log? No need to include an ioreg or the VRMI log.

HeartAndSoul001, 2024-10-12:
here is the log:
[ps2.txt](https://github.com/user-attachments/files/17348838/ps2.txt)

1Revenger1, 2024-10-12:
Hrmm. So shown here are two devices when there should only be one PS2SMBusDevice here.
[image]

Would you mind trying
1) Disabling VoodooSMBus temporarily to see if two services of ApplePS2Synaptics attach? Send an ioreg if they do.
2) Try an older version of VoodooPS2 to see if two services also attach.

From what I am seeing, this is a VoodooPS/2 issue somehow and if we solve that, your touchpad should work.

HeartAndSoul001, 2024-10-12:
@1Revenger1  Yes, you are right. Disabling VoodooSMBus and only one services of ApplePS2Synaptics attach
[image]
[MacBook Pro.zip](https://github.com/user-attachments/files/17348931/MacBook.Pro.zip)

HeartAndSoul001, 2024-10-12:
@1Revenger1 

> [VoodooPS2Controller-2.3.6-RELEASE.zip](https://github.com/user-attachments/files/17349085/VoodooPS2Controller-2.3.6-RELEASE.zip) Mind giving this version of VoodooPS2 a try with VoodooSMBus and co enabled?

yes, it works. thanks a lot for your work!

HeartAndSoul001, 2024-10-12:
> My first thought is that the `BUS0` device isn't going to do anything while VoodooSMBus/VoodooRMI are being used. Maybe try removing that and just leaving the RCFG properties? Unfortunately I can [...]

about YogaSMC, I tried it but got problem.
[image]
[image]

1Revenger1, 2024-10-12:
https://github.com/VoodooSMBus/VoodooRMI/actions/runs/11307144865/artifacts/2048843397
https://github.com/acidanthera/VoodooPS2/actions/runs/11306982958/artifacts/2048810283
Updates to both VoodooPS2 and VoodooSMBus. Mind giving these a shot just to be sure?

For YogaSMC, you do need this SSDT to add a couple of methods. https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/SSDTSample/SSDT-ECRW.dsl
This may be useful too: https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/SSDTSample/SSDT-THINK.dsl

HeartAndSoul001, 2024-10-12:
> https://github.com/VoodooSMBus/VoodooRMI/actions/runs/11307144865/artifacts/2048843397
>
> https://github.com/acidanthera/VoodooPS2/actions/runs/11306982958/artifacts/2048810283
>
> Updates to both [...]

yes，off course, I will try the kext you mentioned above. Do I need to give some logs?

About YogaSMC,I use your work--ECEnabler which has method RE1B,RECB,WE1B,WECB. I mean if I don't need to add SSDT-ECRW.aml which has the same function.Actually, I loaded the both acpi following the YogaSMC README(SSDT-ECRW.aml and SSDT-THINK.aml), but got the problem above. I don't know if the version of YogaSMC has been valid in Sonoma or my wrong config. Do you have a EFI file which use YogaSMC for referring?

1Revenger1, 2024-10-12:
These lines do need to be modified to match the name of your EC in your ACPI: https://github.com/zhen-zen/YogaSMC/blob/master/YogaSMC/SSDTSample/SSDT-ECRW.dsl#L10-L13
YogaSMC still relies on those methods even if ECEnabler is being injected.

No need for logs if the artifacts work.

HeartAndSoul001, 2024-10-13:
> https://github.com/VoodooSMBus/VoodooRMI/actions/runs/11307144865/artifacts/2048843397 https://github.com/acidanthera/VoodooPS2/actions/runs/11306982958/artifacts/2048810283 Updates to both [...]

@1Revenger1 Already test, there are no problem so far

1Revenger1, 2024-10-13:
Thanks! I'll go merge the VoodooPS2 PR in a bit

closed 2024-10-13 by 1Revenger1

HeartAndSoul001, 2024-10-13:
@1Revenger1 Thanks a lot for solving the problem.