VoodooSMBus/VoodooRMI #189: RMIIBus timeout and stall prior to MacOS Login with VoodooRMI 1.4.1

docdude, 2024-10-09:
I have a Haswell Dell Inspiron running Sequoia.  
[image]

Since updating VoodooRMI to 1.4.1 I will either get a RMIIBus timeout just prior to login and will have to reboot, or the boot process stalls just prior to login until I touch the trackpad at which point login screen comes up and then I will no longer have any issues. 

Here is my SSDT that allows the trackpad to work in APIC Interrupt mode. 

DefinitionBlock ("", "SSDT", 2, "hack", "TPD1", 0x00000000)
{
    External (_SB_.PCI0.I2C1.TPD1, DeviceObj)
    External (GR13, FieldUnitObj)

    Scope (_SB.PCI0.I2C1.TPD1)
    {
        Method (_CRS, 0, Serialized)  // _CRS: Current Resource Settings
        {
            Name (SBFB, ResourceTemplate ()
            {
                I2cSerialBusV2 (0x002C, ControllerInitiated, 0x00061A80,
                    AddressingMode7Bit, "\\_SB.PCI0.I2C1",
                    0x00, ResourceConsumer, , Exclusive,
                    )
                Interrupt (ResourceConsumer, Level, **ActiveLow**, Exclusive, ,, _Y00)
                {
                    0x00000027,
                }
            })
            If ((GR13 == One))
            {
                CreateDWordField (SBFB, \_SB.PCI0.I2C1.TPD1._CRS._Y00._INT, VAL3)  // _INT: Interrupts
                VAL3 = 0x1B
            }

            Return (SBFB) /* \_SB_.PCI0.I2C1.TPD1._CRS.SBFB */
        }
    }
}    
Polling mode works by eliminating this from the SSDT:

                Interrupt (ResourceConsumer, Level, **ActiveLow**, Exclusive, ,, _Y00)
                {
                    0x00000027,
                }

GPIO Mode I have not been able to get it to work.  

By changing the Interrupt level to ActiveHigh, MacOS boots without issues but the trackpad isn't functional.  

[    1.385288]: VRMI - Debug: RMII2C::TPD1 probing
[    1.386229]: VRMI - Debug: RMII2C::TPD1 Trying to set mode, attempt 0
[    1.386561]: VRMI - Debug: RMII2C::TPD1 mode set
[    1.888535]: VRMI - Debug: RMII2C::TPD1 powerState 1 : on
[    1.888591]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    1.888598]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    1.888602]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[    1.888606]: VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
[    1.888609]: VRMI - Debug: Config DisableWhileTypingTimeout loaded: 7d0 -> fa
[    1.888614]: VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 7d0 -> fa
[    1.888618]: VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
[    1.888621]: VRMI - Debug: Config ForceTouchType loaded: 1 -> 1
[    1.888624]: VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
[    1.888629]: VRMI - Debug: Config PalmRejectionWidth loaded: f -> a
[    1.888632]: VRMI - Debug: Config PalmRejectionHeight loaded: 50 -> 3c
[    1.888635]: VRMI - Debug: Config PalmRejectionTrackpointHeight loaded: 14 -> 14
[    1.888638]: VRMI - Debug: Updating Configuration
[    1.888657]: VRMI - Info: RMI Bus (DEBUG) Starting up!
[    1.889815]: VRMI - Debug: Function F34 - IRQs: 1 CMD Base: 0x0 CTRL Base: 0x3c DATA Base: 0x0 QRY Base: 0x71
[    1.889820]: VRMI - Info: F34 not implemented
[    1.890953]: VRMI - Debug: Function F1 - IRQs: 1 CMD Base: 0x6f CTRL Base: 0x40 DATA Base: 0x13 QRY Base: 0x7a
[    1.906923]: VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: s3203_ver5, fw id: 1541093
[    1.912099]: VRMI - Debug: Function F11 - IRQs: 2 CMD Base: 0x70 CTRL Base: 0x45 DATA Base: 0x15 QRY Base: 0x97
[    1.930663]: VRMI - Debug: Function F54 - IRQs: 1 CMD Base: 0x173 CTRL Base: 0x10e DATA Base: 0x100 QRY Base: 0x174
[    1.930669]: VRMI - Info: F54 not implemented
[    1.931803]: VRMI - Debug: Function F30 - IRQs: 1 CMD Base: 0x100 CTRL Base: 0x16d DATA Base: 0x10d QRY Base: 0x182
[    1.933986]: VRMI - Debug: F30: Found Button 1 at 2
[    1.937083]: VRMI - Debug: Function F55 - IRQs: 1 CMD Base: 0x201 CTRL Base: 0x200 DATA Base: 0x200 QRY Base: 0x202
[    1.937088]: VRMI - Info: F55 not implemented
[    1.942075]: VRMI - Debug: Setting IRQ Mask: 0x7f Bits: 0x7

There is no more debug output after this with Interrupt level set to ActiveHigh.

As stated with Interrupt level set to ActiveLow I get either RMIIBus timeout and need to reboot or I have to touch the trackpad to boot completely to login screen at which point the trackpad works fine.  

Any insight would be helpful.

1Revenger1, 2024-10-12:
Are you able to grab a boot log from when the touchpad requires user input to boot the system?
`sudo dmesg | grep -iE 'ps2|i2c|vrmi > ~/Desktop/frozen_vrmi_log.txt`. No need to have a debug version of VoodooPS2 or VoodooI2C.

docdude, 2024-10-13:
[frozen_vrmi_log.txt](https://github.com/user-attachments/files/17356532/frozen_vrmi_log.txt)

Here you go.  This problem also on a Dell G5 15 5587 with VoodooRMI 1.4.1.  Same stall and RMIBUS timeout error. 
Again if I touch the trackpad early in boot, boot process will get to login screen otherwise it stalls. 
The log attached is a dmesg dump when the laptop doesn't stall.  Can't get one when it does, I have to reboot.

1Revenger1, 2024-10-13:
Hrmm, are you able to take a picture of where it stalls and shows the RMII2C error? If not, are you able to more specifically write down the errors from RMII2C/VoodooI2C? I don't see anything in that last log you sent unfortunately.

docdude, 2024-10-13:
[image]

1Revenger1, 2024-10-13:
Looks like the image didn't upload

1Revenger1, 2024-10-18:
Can you try this build? This fixed it for someone else with the same symptoms - it comes from the PR linked above. https://github.com/VoodooSMBus/VoodooRMI/actions/runs/11409811472/artifacts/2075818743

docdude, 2024-10-19:
Will load and report

docdude, 2024-10-20:
This is the new log.  This PR loads, no stall prior to login, but the trackpad doesn't work. Only 3 fingers seems to be recognized.
Single finger click works, but can't move the cursor with single finger.

[frozen_vrmi_log.txt](https://github.com/user-attachments/files/17450703/frozen_vrmi_log.txt)

1Revenger1, 2024-10-20:
Can you grab a log using the debug version of VoodooRMI? It should contain more data about the data packets it is getting.
I'll try to see if I can figure out the issue over here.

1Revenger1, 2024-10-20:
[VoodooRMI-1.4.2-DEBUG.zip](https://github.com/user-attachments/files/17450797/VoodooRMI-1.4.2-DEBUG.zip)
I think I see the issue. I had the offset wrong for function 12 contact data in the HID attention reports Want to give this a try?

docdude, 2024-10-21:
Here is the log with Debug kext you posted

Still no single finger movement  but no freeze prior to login

[nosinglefinger_nofreeze_vrmi_log.txt](https://github.com/user-attachments/files/17462568/nosinglefinger_nofreeze_vrmi_log.txt)

1Revenger1, 2024-10-21:
Looks like something wasn't being incremented, want to give this a shot?
[VoodooRMI-1.4.2-DEBUG.zip](https://github.com/user-attachments/files/17463298/VoodooRMI-1.4.2-DEBUG.zip)

docdude, 2024-10-22:
Here is the log with the newest build

[nofreeze_onefinger_works_vrmi_log.txt](https://github.com/user-attachments/files/17469363/nofreeze_onefinger_works_vrmi_log.txt)

Seems to have fixed it.  Will report if anything comes up.  Tested the gestures and they seem to be ok.

PR #193 (2024-10-23): Pass I2C HID input report and timestamp to functions
Closes #189 
Closes #174 
Closes #143
closed 2024-10-23 by 1Revenger1