VoodooSMBus/VoodooRMI #192: SYNA7DAB trackpad doesn't work with VoodooRMI

Idcaboutusernamer, 2024-10-15:
**Describe the bug**
When attempting to load all the kexts required, the trackpad fails to work in macOS. I am using the load order of this, descending order:
VoodooI2C.kext
VoodooI2C.kext/Contents/Plugins/VoodooInput.kext (Disabled)
VoodooI2C.kext/Contents/Plugins/VoodooI2CServices.kext
VoodooI2C.kext/Contents/Plugins/VoodooGPIO.kext
VoodooRMI.kext
VoodooRMI.kext/Contents/Plugins/RMII2C.kext
VoodooRMI.kext/Contents/Plugins/VoodooInput.kext
VoodooI2CHID.kext
VoodooPS2Controller.kext
VoodooPS2Controller.kext/Contents/Plugins/VoodooInput.kext (Disabled)
VoodooPS2Controller.kext/Contents/Plugins/VoodooPS2Trackpad.kext
VoodooPS2Controller.kext/Contents/Plugins/VoodooPS2Keyboard.kext
VoodooPS2Controller.kext/Contents/Plugins/VoodooPS2Mouse.kext
My laptop is a Acer Spin SP314-54N with a Intel Core i5-1035G1, running Sonoma 14.4.1

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to load VoodooRMI on the SYNA7DAB trackpad, with the load order specified
2. Reboot
3. Trackpad isn't working

**Expected behavior**
The expected behaviour is that the trackpad works instead of not working or responding.

**Log**
I attempted to make a lot of VoodooRMI but both ways outlined in the readme came out empty, I am not sure why as the kext shows up as loaded in kextstat. Here is my IOReg exported from IORegistryExplorer: https://drive.google.com/file/d/1Jhq97zr2ffoYdBiY7vo2EC1go41ESx26/view?usp=drivesdk

Thanks all for any help or guidance

1Revenger1, 2024-10-15:
Looks like an issue with VoodooI2C. I don't see it attaching (but I do see VoodooI2CServices elsewhere in the ioreg so it is injected at least). I am going to close this issue as it is unrelated to VoodooRMI.

`sudo dmesg` works better if you have both DebugEnhancer.kext and `msgbuf=1048576` in your boot-args. You should be able to see logs after that. Maybe try filtering with `sudo dmesg | grep -iE 'i2c|vrmi|ps2'`. You will want to disable VoodooPS2Trackpad and Mouse it looks like since those both are attaching for some reason.

closed 2024-10-15 by 1Revenger1

Idcaboutusernamer, 2024-10-15:
So this is the log I got after doing the steps you recommended:
[    1.338513]: ApplePS2Controller: Notification consumer published: ApplePS2Controller
[    1.348899]: ApplePS2Controller: Notification consumer published: ApplePS2Keyboard
[   10.639210]: Lilu   patcher: @ failed to solve __ZN31AppleIntelFramebufferController14ReadI2COverAUXEP21AppleIntelFramebufferP21AppleIntelDisplayPathjtPhbh, err 2
[   10.639213]: WhateverGreen      igfx: @ I2C: Failed to route functions.
[   10.901856]: /System/Library/DriverExtensions/com.apple.AppleUserHIDDrivers.dext/com.apple.AppleUserHIDDrivers[323] ==> com.apple.dextBoot args: msgbuf=1048576 keepsyms=1 swd_panic=1 -lilubetaall -vi2c-force-polling watchdog=0 agdpmod=vit9696 -noDC9 TSC_sync_margin=0 plugin-type=1
[   10.939069]: Boot args: msgbuf=1048576 keepsyms=1 swd_panic=1 -lilubetaall -vi2c-force-polling watchdog=0 agdpmod=vit9696 -noDC9 TSC_sync_margin=0 plugin-type=1
[   23.188379]: /System/Library/DriverExtensions/IOUserBluetoothSerialDriver.dext/IOUserBluetoothSerialDriver[372] ==> com.apple.dextBoot args: msgbuf=1048576 keepsyms=1 swd_panic=1 -lilubetaall -vi2c-force-polling watchdog=0 agdpmod=vit9696 -noDC9 TSC_sync_margin=0 plugin-type=1