VoodooSMBus/VoodooRMI #195: SYNA3091 troubles

P11x3l, 2025-02-06:
Hi Guys, 

I'm working on a HP Elitebook 840 G6 with touchpoint & trackpad SYNA3091. Touchpoint is ps2 and working fine but the trackpad is giving me headaches!

VoodooI2C seems to find the device OK and RMI seems to hook F12, There's a trackpad showing up under system settings but when you try to use the trackpad the pointer is unresponsive to mouse movements or clicks.

Any tips please?

```
~ % sudo dmesg | grep -i vrmi  
[    1.645369]: VRMI - Debug: RMII2C::TPD0 probing
[    1.646374]: VRMI - Debug: RMII2C::TPD0 Trying to set mode, attempt 0
[    1.646725]: VRMI - Debug: RMII2C::TPD0 mode set
[    2.148138]: VRMI - Debug: RMII2C::TPD0 powerState 1 : on
[    2.148202]: VRMI - Debug: Config TrackpointMultiplier loaded: a -> a
[    2.148207]: VRMI - Debug: Config TrackpointScrollMultiplierX loaded: a -> 1e
[    2.148210]: VRMI - Debug: Config TrackpointScrollMultiplierY loaded: a -> 1e
[    2.148212]: VRMI - Debug: Config TrackpointDeadzone loaded: 1 -> 1
[    2.148215]: VRMI - Debug: Config DisableWhileTypingTimeout loaded: 7d0 -> fa
[    2.148218]: VRMI - Debug: Config DisableWhileTrackpointTimeout loaded: 7d0 -> fa
[    2.148220]: VRMI - Debug: Config ForceTouchMinPressure loaded: 50 -> 5a
[    2.148222]: VRMI - Debug: Config ForceTouchType loaded: 1 -> 1
[    2.148225]: VRMI - Debug: Config MinYDiffThumbDetection loaded: c8 -> c8
[    2.148228]: VRMI - Debug: Config PalmRejectionWidth loaded: f -> a
[    2.148230]: VRMI - Debug: Config PalmRejectionHeight loaded: 50 -> 3c
[    2.148232]: VRMI - Debug: Config PalmRejectionTrackpointHeight loaded: 14 -> 14
[    2.148234]: VRMI - Debug: Updating Configuration
[    2.148248]: VRMI - Info: RMI Bus (DEBUG) Starting up!
[    2.148952]: VRMI - Debug: Function F34 - IRQs: 1 CMD Base: 0x0 CTRL Base: 0x11 DATA Base: 0x0 QRY Base: 0x26
[    2.148955]: VRMI - Info: F34 not implemented
[    2.149631]: VRMI - Debug: Function F1 - IRQs: 2 CMD Base: 0x25 CTRL Base: 0x12 DATA Base: 0x6 QRY Base: 0x2e
[    2.154950]: VRMI - Info: Found RMI4 device, manufacturer: Synaptics, product: TM335, fw id: 3170022
[    2.157647]: VRMI - Debug: Function F12 - IRQs: 2 CMD Base: 0x0 CTRL Base: 0x1a DATA Base: 0xb QRY Base: 0x4e
[    2.161530]: VRMI - Debug: F12 - reg: 0 reg size: 1 subpackets: 1
[    2.161533]: VRMI - Debug: F12 - reg: 1 reg size: 1 subpackets: 1
[    2.161535]: VRMI - Debug: F12 - reg: 2 reg size: 3 subpackets: 2
[    2.161537]: VRMI - Debug: F12 - reg: 3 reg size: 42 subpackets: 12
[    2.161539]: VRMI - Debug: F12 - reg: 4 reg size: 1 subpackets: 1
[    2.161541]: VRMI - Debug: F12 - reg: 5 reg size: 12 subpackets: 2
[    2.161543]: VRMI - Debug: F12 - reg: 6 reg size: 26 subpackets: 0
[    2.161545]: VRMI - Debug: F12 - reg: 7 reg size: 0 subpackets: 0
[    2.161547]: VRMI - Debug: F12 - reg: 8 reg size: 0 subpackets: 0
[    2.161549]: VRMI - Debug: F12 - reg: 9 reg size: 0 subpackets: 0
[    2.161551]: VRMI - Debug: F12 - reg: 10 reg size: 0 subpackets: 0
[    2.161553]: VRMI - Debug: F12 - reg: 13 reg size: 0 subpackets: 0
[    2.164327]: VRMI - Debug: F12 - reg: 8 reg size: 15 subpackets: 5
[    2.164329]: VRMI - Debug: F12 - reg: 9 reg size: 20 subpackets: 5
[    2.164331]: VRMI - Debug: F12 - reg: 10 reg size: 19 subpackets: 5
[    2.164333]: VRMI - Debug: F12 - reg: 11 reg size: 22 subpackets: 6
[    2.164335]: VRMI - Debug: F12 - reg: 15 reg size: 7 subpackets: 3
[    2.164337]: VRMI - Debug: F12 - reg: 19 reg size: 67 subpackets: 5
[    2.164339]: VRMI - Debug: F12 - reg: 20 reg size: 0 subpackets: 0
[    2.164341]: VRMI - Debug: F12 - reg: 22 reg size: 0 subpackets: 0
[    2.164343]: VRMI - Debug: F12 - reg: 23 reg size: 0 subpackets: 0
[    2.164345]: VRMI - Debug: F12 - reg: 28 reg size: 0 subpackets: 0
[    2.164347]: VRMI - Debug: F12 - reg: 69 reg size: 0 subpackets: 0
[    2.166420]: VRMI - Debug: F12 - reg: 1 reg size: 40 subpackets: 5
[    2.166422]: VRMI - Debug: F12 - reg: 2 reg size: 1 subpackets: 1
[    2.166424]: VRMI - Debug: F12 - reg: 4 reg size: 1 subpackets: 1
[    2.166426]: VRMI - Debug: F12 - reg: 5 reg size: 15 subpackets: 5
[    2.166428]: VRMI - Debug: F12 - reg: 9 reg size: 10 subpackets: 5
[    2.166430]: VRMI - Debug: F12 - reg: 15 reg size: 1 subpackets: 1
[    2.166432]: VRMI - Debug: F12 - Data packet size: 0x44
[    2.167436]: VRMI - Debug: F12 - Number of fingers 5
[    2.169080]: VRMI - Debug: Function F54 - IRQs: 1 CMD Base: 0x13d CTRL Base: 0x110 DATA Base: 0x100 QRY Base: 0x13e
[    2.169083]: VRMI - Info: F54 not implemented
[    2.170695]: VRMI - Debug: Function F30 - IRQs: 1 CMD Base: 0x200 CTRL Base: 0x201 DATA Base: 0x200 QRY Base: 0x207
[    2.171934]: VRMI - Debug: F30: Found Button 1 at 2
[    2.173625]: VRMI - Debug: Function F55 - IRQs: 1 CMD Base: 0x308 CTRL Base: 0x300 DATA Base: 0x300 QRY Base: 0x309
[    2.173628]: VRMI - Info: F55 not implemented
[... 57 lines cut ...]
[    6.774353]: VoodooInputActuatorDevice:0x100000446 start
[    6.774812]: VoodooInputActuatorDevice:0x100000446 creating interfaces
[    6.774833]: IOHIDInterface:0x10000044b start for VoodooInputActuatorDevice:0x100000446
[    6.777788]: VoodooInputSimulatorDevice:0x100000411 open by AppleMultitouchTrackpadHIDEventDriver:0x10000044d (0x0)
[    6.778536]: VoodooInputActuatorDevice:0x100000446 open by AppleActuatorHIDEventDriver:0x100000450 (0x0)
[   17.604335]: VoodooInputSimulatorDevice:0x100000359 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[   17.604409]: VoodooInputSimulatorDevice:0x100000411 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[   17.738262]: VoodooInputSimulatorDevice:0x100000411 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[   17.738283]: VoodooInputSimulatorDevice:0x100000359 Matching has vendor DeviceUsagePage : ff0c bundleIdentifier com.apple.AppleUserHIDDrivers ioclass AppleUserHIDEventService but transport and vendorID is missing
[  637.557164]: VoodooGPIOCannonLakeLP::Going to Sleep!
[  637.557851]: VoodooI2CControllerDriver::pci8086,9de8 Going to sleep
[  637.559097]: VoodooI2CPCIController::pci8086,9de8 Going to sleep
[  637.583428]: VoodooI2CControllerDriver::pci8086,9de9 Going to sleep
[  637.584773]: VoodooI2CPCIController::pci8086,9de9 Going to sleep
[  651.282144]: VoodooGPIOCannonLakeLP::Woke up from Sleep!
[  651.300720]: VoodooI2CPCIController::pci8086,9de9 Set PCI power state D0
[  651.300736]: VoodooI2CPCIController::pci8086,9de9 Woke up
[  651.301441]: VoodooI2CPCIController::pci8086,9de8 Set PCI power state D0
[  651.301453]: VoodooI2CPCIController::pci8086,9de8 Woke up
[  651.301459]: VoodooI2CControllerDriver::pci8086,9de9 Woke up
[  651.301529]: VoodooI2CControllerDriver::pci8086,9de8 Woke up

```

[image]

1Revenger1, 2025-02-06:
Can you grab the logs from VoodooPS2 as well? It looks like the PS/2 driver is attaching and I wonder if that is causing issues.
Can you double check `sudo powermetrics` to see if there are interrupts from either the GPIO or I2C device?

It will be a few days before I can dig into this but I think those are a couple good first steps.
Couple other thoughts for myself really:
- HID reset needed possibly?

P11x3l, 2025-02-07:
Thanks 1Revenger1,  I'm at the limits of my troubleshooting skills for this one so appreciative of any help!

The trackpad does actually work with only voodoops2 loaded (i2c and rmi kexts removed) but it's super laggy and doesn't register clicks properly. I can't see anything in the powermetrics log about GPIO or I2C but attached anyway.

[powermetrics.txt](https://github.com/user-attachments/files/18702177/powermetrics.txt)
[vps2.txt](https://github.com/user-attachments/files/18702178/vps2.txt)

P11x3l, 2025-02-08:
Disabled VoodooPS2Mouse.kext & VoodooPS2Trackpad.kext, as expected all mouse functionality stopped.

VRMI logs look almost the same, my original logs had a couple of lines about power states at the bottom which are no longer present.

[grep ps2.txt](https://github.com/user-attachments/files/18716049/grep.ps2.txt)
[image]
[grep voodoo.txt](https://github.com/user-attachments/files/18716048/grep.voodoo.txt)
[grep vrmi.txt](https://github.com/user-attachments/files/18716050/grep.vrmi.txt)

1Revenger1, 2025-02-08:
Hrmm, do you have an SSDT to set the GPIO pin or is this one assigned by the firmware?

P11x3l, 2025-02-08:
I haven't done any custom patching, so it must be set by firmware currently

1Revenger1, 2025-02-08:
Got it, can you try adding `-vi2c-force-polling` real quick?

P11x3l, 2025-02-09:
Very cool, some progress! VRMI log is showing finger detections and gestures etc are working! Googling `-vi2c-force-polling` has led me to a section of the Voodooi2c manual on GPIO interrupts and pinning, guess that's my next port of call.

Thanks for your help :)

[vrmi.txt](https://github.com/user-attachments/files/18723023/vrmi.txt)

closed 2025-04-24 by 1Revenger1